ARM-DOE/pyart #230: Segfault with dealias_unwrap_phase

kirknorth, 2015-01-21:
@jjhelmus, I believe `dealias_unwrap_phase` segfaults when `unwrap_unit` is set to <i>sweep</i> and there are sweeps with no data, e.g. some of the first few tilts for certain NEXRAD scan strategies, like this [file](https://www.dropbox.com/s/y85esnc2i4lfj0x/KINX20110425_091822_V03?dl=0). To test this, I filled the radial velocity field with a constant value of 10 m s<sup>-1</sup>, and the routine completed.

jjhelmus, 2015-01-21:
Verified.  The issue comes from the unwrap_2d_ljmu.c file.  In the **unwrap2D** function no check is made that the number of edges (params.no_of_edges) is not 0 (or 1) before sorting the edges based on their reliability.   Since this file came from the scikit-image project I want to fix this upstream before fixing it in Py-ART.  Should have this fixed later this week.

jjhelmus, 2015-01-21:
BTW, @kirknorth, thanks for catching this.  I'll have a third dealiasing routine for you to play with soon.

kirknorth, 2015-01-21:
Sounds good :+1:

PR #231 (2015-01-21): BUG: No SegFault when dealiasing sweeps with all gates filtered
The dealias_unwrap_phase function no longer Segmentation faults when
a sweep in the passed radar object is fully masked or all the gates
are filtered.

Closes #230
commit 32cb5f39 (2015-01-21): BUG: No SegFault when dealiasing sweeps with all gates filtered

The dealias_unwrap_phase function no longer Segmentation faults when
a sweep in the passed radar object is fully masked or all the gates
are filtered.

Closes #230
closed 2015-01-21 by jjhelmus