ARM-DOE/pyart #239: undefined symbol

rvalenzuelar, 2015-02-05:
I'm getting the following message when I try to import pyart:

```python
In [2]: import pyart
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-2-706b47c96bb9> in <module>()
----> 1 import pyart

/usr/local/lib/python2.7/dist-packages/pyart/__init__.py in <module>()
     11 # import subpackages
     12 from . import core
---> 13 from . import io
     14 from . import correct
     15 from . import graph

/usr/local/lib/python2.7/dist-packages/pyart/io/__init__.py in <module>()
     32     _RSL_AVAILABLE = False
     33 from .mdv import read_mdv
---> 34 from .sigmet import read_sigmet
     35 from .chl import read_chl
     36 from .cfradial import read_cfradial, write_cfradial

/usr/local/lib/python2.7/dist-packages/pyart/io/sigmet.py in <module>()
     24 from ..core.radar import Radar
     25 from .common import make_time_unit_str
---> 26 from ._sigmetfile import SigmetFile, bin4_to_angle, bin2_to_angle
     27 from . import _sigmet_noaa_hh
     28 

ImportError: /usr/local/lib/python2.7/dist-packages/pyart/io/_sigmetfile.so: undefined symbol: PyUnicodeUCS2_DecodeUTF8

```

jjhelmus, 2015-02-05:
How did you install Py-ART?  The error indicates that the C extensions in Py-ART were build against a Python interpreter that uses a different unicode encoding standard than the interpreter that you are using.

rvalenzuelar, 2015-02-05:
I cloned the Github repo and then followed the installation instructions. I have pyart installed in:

```bash
~/Enthought/Canopy_32bit/User/lib/python2.7/site-packages/pyart
```
and
```bash
/usr/local/lib/python2.7/dist-packages/pyart
```
The error message shows up when I try to import using ipython in a terminal window. If I import using Canopy I get no error message.

jjhelmus, 2015-02-05:
Try calling the Enthought's version of IPython not the system IPython.  `which ipython` will provide the full path to the binary.  It should be something like ~/Enthought/Canopy_32bit/User/bin/ipython, if it is not, you will want to adjust your PATH variable so that the Enthought IPython is picked up before the system Python.

The longer explanation is:

You have two Python installs on your machine:
* The system Python, likely in /usr/bin/python or /usr/local/bin/python
* The Enthough Canopy Python, I'm guessing in ~/Enthought/Canopy_32bit/User/bin/python

From the two paths in look like Py-ART got installed for both Python installs (the system Python and Enthought Python).  The system and Enthough Python interpreters use different representations for for unicode which are incompatible, extensions compiled against one version will not work with the other.  This leads to issues when the two interpreters and libraries are mixed.

rvalenzuelar, 2015-02-16:
I tried to "isolate" the installation by using conda and the recipie located in the pyart repo but I get the same error:

```bash
===== testing package: pyart-git-np19py27_0 =====
import: u'pyart'
Traceback (most recent call last):
  File "/home/raul/miniconda/conda-bld/test-tmp_dir/run_test.py", line 25, in <module>
    import pyart
  File "/home/raul/miniconda/envs/_test/lib/python2.7/site-packages/pyart/__init__.py", line 13, in <module>
    from . import io
  File "/home/raul/miniconda/envs/_test/lib/python2.7/site-packages/pyart/io/__init__.py", line 34, in <module>
    from .sigmet import read_sigmet
  File "/home/raul/miniconda/envs/_test/lib/python2.7/site-packages/pyart/io/sigmet.py", line 26, in <module>
    from ._sigmetfile import SigmetFile, bin4_to_angle, bin2_to_angle
ImportError: /home/raul/miniconda/envs/_test/lib/python2.7/site-packages/pyart/io/_sigmetfile.so: undefined symbol: PyUnicodeUCS2_DecodeUTF8
TESTS FAILED: pyart-git-np19py27_0
```
Should not be the installation interpreter-independent? It does not make really sense to me that I can use pyart only with the canopy version of python. Should I uninstall canopy? or the system python? (I'm looking to work with miniconda from now on)

closed 2015-02-16 by rvalenzuelar

jjhelmus, 2015-02-16:
Pure python modules (those which only contain Python code and not compiled extensions) are interpreter-independent.  Compiled extensions, such as _sigmefile, link against libpython are therefore may become interpreter dependent if the Python interpreters were compiled differently. For example if different Unicode support schemes are used, UCS2 vs UCS4, then the module can only be used with interpreter against which the module was linked.

So yes, I expect that a version of Py-ART compiled against Canopy may not work when used with the system Python.  Using only a single Python interpreter is recommended,but uninstalling the other interpreter is not need, removing it from the PATH should suffice.

A good method of checking which interpreter Py-ART, or another compiled modules, is compiled against is to see which libpython the module is linked against using the **ldd**.  If the module is linked against the libpython in /usr/lib or /usr/local/lib then the module is compiled against the system Python.  If the libpython library is in an Anaconda or Canopy directory then it is compiled against Continuum's or Enthoughts' Python.

For example here is the results from a Py-ART _sigmetfile compiled against Anaconda's libpython:

```
~$ ldd /home/jhelmus/anaconda/envs/sandbox/lib/python2.7/site-packages/pyart/io/_sigmetfile.so
	linux-vdso.so.1 =>  (0x00007fff569fe000)
	libpython2.7.so.1.0 => /home/jhelmus/anaconda/envs/sandbox/lib/python2.7/site-packages/pyart/io/../../../../libpython2.7.so.1.0 (0x00007f4ac358e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4ac3354000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4ac2f95000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f4ac2d91000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f4ac2b8d000)
	libm.so.6 => /home/jhelmus/anaconda/envs/sandbox/lib/python2.7/site-packages/pyart/io/../../../.././libm.so.6 (0x00007f4ac290a000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f4ac3bef000)
```

And here is one compiled against the system Python:

```
~$ ldd ~/python/pyart/pyart/io/_sigmetfile.so 
	linux-vdso.so.1 =>  (0x00007fff0c505000)
	libpython2.7.so.1.0 => /usr/lib/libpython2.7.so.1.0 (0x00007f18434c1000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f18432a4000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1842ee4000)
	libssl.so.1.0.0 => /lib/x86_64-linux-gnu/libssl.so.1.0.0 (0x00007f1842c86000)
	libcrypto.so.1.0.0 => /lib/x86_64-linux-gnu/libcrypto.so.1.0.0 (0x00007f18428ab000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f1842693000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f184248f000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f184228c000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1841f8f000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f1841d79000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1843c6c000)
```

I my system (Ubuntu 12.04) both of these work as both the system Python and the version of Anaconda I installed support UCS4 Unicode.  If I had an interpreter which had UCS2 support then these modules would not work.  You can check which Unicode support scheme a Python interpreter support using  `python -c "import sys; print(sys.maxunicode)"`.  If that command print 111411 then Python supports UCS4, if 65535 then UCS2.

rvalenzuelar, 2015-02-16:
I checked the python sys.maxunicode and I got:

- Miniconda python (e.g. /home/raul/miniconda/bin/python): UCS4
- System python (e.g. /usr/bin/python): UCS4
- Canopy python (e.g. /home/raul/Enthought/Canopy_32bit/User/bin/python): UCS2

Then, I <code>ldd</code> _sigmetfile.so and I get:

Miniconda:
```bash
raul@raul-ubuntu:~/miniconda/envs/_test/lib/python2.7/site-packages/pyart/io$ ldd _sigmetfile.so 
	linux-gate.so.1 =>  (0xb76e5000)
	libpython2.7.so.1.0 => /home/raul/miniconda/envs/_test/lib/python2.7/site-packages/pyart/io/./../../../../libpython2.7.so.1.0 (0xb74ce000)
	libpthread.so.0 => /lib/i386-linux-gnu/libpthread.so.0 (0xb7498000)
	libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xb72e8000)
	libdl.so.2 => /lib/i386-linux-gnu/libdl.so.2 (0xb72e3000)
	libutil.so.1 => /lib/i386-linux-gnu/libutil.so.1 (0xb72df000)
	libm.so.6 => /home/raul/miniconda/envs/_test/lib/python2.7/site-packages/pyart/io/./../../../.././libm.so.6 (0xb72b6000)
	/lib/ld-linux.so.2 (0xb76e6000)
```
System:
```bash
raul@raul-ubuntu:/usr/local/lib/python2.7/dist-packages/pyart/io$ ldd _sigmetfile.so 
	linux-gate.so.1 =>  (0xb7776000)
	libpython2.7.so.1.0 => /usr/lib/i386-linux-gnu/libpython2.7.so.1.0 (0xb73a0000)
	libpthread.so.0 => /lib/i386-linux-gnu/libpthread.so.0 (0xb7384000)
	libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xb71d4000)
	libz.so.1 => /lib/i386-linux-gnu/libz.so.1 (0xb71ba000)
	libdl.so.2 => /lib/i386-linux-gnu/libdl.so.2 (0xb71b5000)
	libutil.so.1 => /lib/i386-linux-gnu/libutil.so.1 (0xb71b1000)
	libm.so.6 => /lib/i386-linux-gnu/libm.so.6 (0xb716b000)
	/lib/ld-linux.so.2 (0xb7777000)
```
and Canopy:
```bash
raul@raul-ubuntu:~/Enthought/Canopy_32bit/User/lib/python2.7/site-packages/pyart/io$ ldd _sigmetfile.so 
	linux-gate.so.1 =>  (0xb77dd000)
	libpython2.7.so.1.0 => /usr/lib/i386-linux-gnu/libpython2.7.so.1.0 (0xb7407000)
	libpthread.so.0 => /lib/i386-linux-gnu/libpthread.so.0 (0xb73eb000)
	libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xb723b000)
	libz.so.1 => /lib/i386-linux-gnu/libz.so.1 (0xb7221000)
	libdl.so.2 => /lib/i386-linux-gnu/libdl.so.2 (0xb721c000)
	libutil.so.1 => /lib/i386-linux-gnu/libutil.so.1 (0xb7218000)
	libm.so.6 => /lib/i386-linux-gnu/libm.so.6 (0xb71d2000)
	/lib/ld-linux.so.2 (0xb77de000)
```

Here is a point that still confuses me (I suspect I'm missing something here):

-Miniconda python (UCS4) runs the test using _sigmetfile.so compiled with the Miniconda python (UCS4)
-System python (UCS4) uses _sigmetfile.so compiled with system Python (UCS4)
-Canopy python (UCS2) uses _sigmetfile.so compiled with system Python (UCS4) 

From this result I would expect to Canopy python to fail when calling pyart; nevertheless, the behavior is the following:

- With Miniconda python (when building conda recipe): <code>undefined symbol: PyUnicodeUCS2_DecodeUTF8</code>
- With System python: import pyart -->  <code>undefined symbol: PyUnicodeUCS2_DecodeUTF8</code>
- With Canopy python: import pyart --> success (some warnings about plotting support show up)

So, if all <code>_sigmetfile.so</code> were compiled against a UCS4 libpython, why are they failing to compile with the UCS4 interpreters? Also, the error seems contradictory because is telling me that the module (sigmetfile) is UCS2 but <code>ldd</code> indicates that they are UCS4.

jjhelmus, 2015-02-16:
That does seem strange, I agree I would expect the opposite behavior.  A couple other thoughts:

Is the LD_LIBRARY_PATH variable set (check with the `env` command)?  

Perhaps at compile time the incorrect version of Python.h is used for definitions which are not present in the linked library.  You can dump symbols from a library using `nm _sigmetfile.so`.  See if the _sigmetfile.so libraries have the PyUnicodeUCS2_DecodeUTF8 or UCS4 symbol in them using something like:
`nm _sigmetfile.so | grep "PyUnicode"`

If they have the UCS2 symbols but are linked to linked to a UCS4 libpython library (or vice-versa) then something is going wrong during compilation and linking.  The Unicode functions are defined in unicodeobject.h which is included from Python.h.  Watch during compilation which Python.h file is included, it might be the wrong one.  Perhaps Canopy is setting the search path so that its Python.h file is found instead of the system or Anaconda version.

rvalenzuelar, 2015-02-17:
I have <code> LD_LIBRARY_PATH=/usr/local/lib</code>

I dumped the symbols. The three _sigmetfile.so have the PyUnicodeUCS2_DecodeUTF8 symbol. This makes sense since Canopy python is the only interpreter that give no error message.

I could not find any reference to <code>Python.h</code> during the pyart building. When finished, I went to check the build directory. I found _sigmetfile.so and after <code>ldd</code>:
```bash
raul@raul-ubuntu:~/Github/pyart/build/lib.linux-i686-2.7/pyart/io$ ldd _sigmetfile.so 
	linux-gate.so.1 =>  (0xb779a000)
	libpython2.7.so.1.0 => /usr/lib/i386-linux-gnu/libpython2.7.so.1.0 (0xb73bc000)
	libpthread.so.0 => /lib/i386-linux-gnu/libpthread.so.0 (0xb73a0000)
	libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xb71f0000)
	libz.so.1 => /lib/i386-linux-gnu/libz.so.1 (0xb71d6000)
	libdl.so.2 => /lib/i386-linux-gnu/libdl.so.2 (0xb71d1000)
	libutil.so.1 => /lib/i386-linux-gnu/libutil.so.1 (0xb71cd000)
	libm.so.6 => /lib/i386-linux-gnu/libm.so.6 (0xb7187000)
	/lib/ld-linux.so.2 (0xb779b000)
```
So it keeps saying that I'm using libpython from <code>/usr/lib/</code>. 

Below is the log for the building with Canopy python. I could not find any reference to <code>/usr/lib/</code> either. I'm not sure what else to do :(
```bash
running build
running config_cc
unifing config_cc, config, build_clib, build_ext, build commands --compiler options
running config_fc
unifing config_fc, config, build_clib, build_ext, build commands --fcompiler options
running build_src
build_src
building extension "pyart.io._sigmetfile" sources
building extension "pyart.__check_build._check_build" sources
building extension "pyart.correct._unwrap_1d" sources
building extension "pyart.correct._unwrap_2d" sources
building extension "pyart.correct._unwrap_3d" sources
building extension "pyart.map.ball_tree" sources
building extension "pyart.map.ckdtree" sources
building extension "pyart.map._load_nn_field_data" sources
building extension "pyart.retrieve._echo_steiner" sources
creating build
creating build/src.linux-i686-2.7
creating build/src.linux-i686-2.7/pyart
creating build/src.linux-i686-2.7/pyart/retrieve
customize Gnu95FCompiler
Found executable /usr/bin/gfortran
customize Gnu95FCompiler
customize Gnu95FCompiler using config
Fortran f77 compiler: /usr/bin/gfortran -Wall -ffixed-form -fno-second-underscore -fPIC -O3 -funroll-loops
Fortran f90 compiler: /usr/bin/gfortran -Wall -fno-second-underscore -fPIC -O3 -funroll-loops
Fortran fix compiler: /usr/bin/gfortran -Wall -ffixed-form -fno-second-underscore -Wall -fno-second-underscore -fPIC -O3 -funroll-loops
compile options: '-c'
gfortran:fix: _configtest.f90
success!
removing: _configtest.f90 _configtest.o
  adding 'pyart/retrieve/_echo_steiner.pyf' to sources.
  adding 'pyart/retrieve/src/echo_steiner.f90' to sources.
f2py options: []
f2py: pyart/retrieve/_echo_steiner.pyf
Reading fortran codes...
	Reading file 'pyart/retrieve/_echo_steiner.pyf' (format:free)
Post-processing...
	Block: _echo_steiner
			Block: convective_radius
			Block: peakedness
			Block: classify
Post-processing (stage 2)...
Building modules...
	Building module "_echo_steiner"...
		Constructing wrapper function "convective_radius"...
		  conv_rad = convective_radius(ze_bkg,area_relation)
		Constructing wrapper function "peakedness"...
		  peak = peakedness(ze_bkg,peak_relation)
[... 209 lines cut ...]
                 from build/src.linux-i686-2.7/fortranobject.c:2:
/home/raul/Enthought/Canopy_32bit/User/lib/python2.7/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^
compiling Fortran sources
Fortran f77 compiler: /usr/bin/gfortran -Wall -ffixed-form -fno-second-underscore -fPIC -O3 -funroll-loops
Fortran f90 compiler: /usr/bin/gfortran -Wall -fno-second-underscore -fPIC -O3 -funroll-loops
Fortran fix compiler: /usr/bin/gfortran -Wall -ffixed-form -fno-second-underscore -Wall -fno-second-underscore -fPIC -O3 -funroll-loops
creating build/temp.linux-i686-2.7/pyart/retrieve
creating build/temp.linux-i686-2.7/pyart/retrieve/src
compile options: '-Ibuild/src.linux-i686-2.7 -I/home/raul/Enthought/Canopy_32bit/User/lib/python2.7/site-packages/numpy/core/include -I/home/raul/Canopy/appdata/canopy-1.5.1.2730.rh5-x86/include/python2.7 -c'
gfortran:f90: pyart/retrieve/src/echo_steiner.f90
/usr/bin/gfortran -Wall -Wall -shared build/temp.linux-i686-2.7/build/src.linux-i686-2.7/pyart/retrieve/_echo_steinermodule.o build/temp.linux-i686-2.7/build/src.linux-i686-2.7/fortranobject.o build/temp.linux-i686-2.7/pyart/retrieve/src/echo_steiner.o -L/home/raul/Canopy/appdata/canopy-1.5.1.2730.rh5-x86/lib -lpython2.7 -lgfortran -o build/lib.linux-i686-2.7/pyart/retrieve/_echo_steiner.so
running build_scripts
creating build/scripts.linux-i686-2.7
copying and adjusting scripts/anytocfradial -> build/scripts.linux-i686-2.7
copying and adjusting scripts/radar_plot -> build/scripts.linux-i686-2.7
copying and adjusting scripts/check_cfradial -> build/scripts.linux-i686-2.7
copying and adjusting scripts/radar_info -> build/scripts.linux-i686-2.7
changing mode of build/scripts.linux-i686-2.7/anytocfradial from 664 to 775
changing mode of build/scripts.linux-i686-2.7/radar_plot from 664 to 775
changing mode of build/scripts.linux-i686-2.7/check_cfradial from 664 to 775
changing mode of build/scripts.linux-i686-2.7/radar_info from 664 to 775

```

jjhelmus, 2015-02-17:
The definition and eventually the symbols come from the **Python.h** file which is getting picked up in the `-I/home/raul/Canopy/appdata/canopy-1.5.1.2730.rh5-x86/include/python2.7` compile option.  My guess it that you will see a similar include statement when you compile using the system or Anaconda based Python which is the root cause of the issue.   Can you post a build log for one of those to verify?

Perhaps Canopy is setting an environment CFLAG or C_INCLUDE_PATH in .bashrc or .bash_profile?  Can you post the results of `env`?

rvalenzuelar, 2015-02-27:
Ok, Anaconda and systems python are compiling with `-I/usr/include/python2.7` (attached below are the build logs).

The only paths that were added to my .bashrc file are these:
```bash
# Added by Canopy installer on 2014-12-15
# VIRTUAL_ENV_DISABLE_PROMPT can be set to '' to make bashprompt show that Canopy is active, otherwise 1
# VIRTUAL_ENV_DISABLE_PROMPT=1 source /home/raul/Enthought/Canopy_32bit/User/bin/activate

# added by Miniconda 3.8.3 installer
export PATH="/home/raul/miniconda/bin:$PATH"
```
I comment/uncomment them depending on what interpreter I want to use.

Here is the log for Anaconda:
```bash
running build
running config_cc
unifing config_cc, config, build_clib, build_ext, build commands --compiler options
running config_fc
unifing config_fc, config, build_clib, build_ext, build commands --fcompiler options
running build_src
build_src
building extension "pyart.io._sigmetfile" sources
building extension "pyart.__check_build._check_build" sources
building extension "pyart.correct._unwrap_1d" sources
building extension "pyart.correct._unwrap_2d" sources
building extension "pyart.correct._unwrap_3d" sources
building extension "pyart.map.ball_tree" sources
building extension "pyart.map.ckdtree" sources
building extension "pyart.map._load_nn_field_data" sources
building extension "pyart.retrieve._echo_steiner" sources
creating build
creating build/src.linux-i686-2.7
creating build/src.linux-i686-2.7/pyart
creating build/src.linux-i686-2.7/pyart/retrieve
customize Gnu95FCompiler
Found executable /usr/bin/gfortran
customize Gnu95FCompiler
customize Gnu95FCompiler using config
Fortran f77 compiler: /usr/bin/gfortran -Wall -ffixed-form -fno-second-underscore -fPIC -O3 -funroll-loops
Fortran f90 compiler: /usr/bin/gfortran -Wall -fno-second-underscore -fPIC -O3 -funroll-loops
Fortran fix compiler: /usr/bin/gfortran -Wall -ffixed-form -fno-second-underscore -Wall -fno-second-underscore -fPIC -O3 -funroll-loops
compile options: '-c'
gfortran:fix: _configtest.f90
success!
removing: _configtest.f90 _configtest.o
  adding 'pyart/retrieve/_echo_steiner.pyf' to sources.
  adding 'pyart/retrieve/src/echo_steiner.f90' to sources.
f2py options: []
f2py: pyart/retrieve/_echo_steiner.pyf
Reading fortran codes...
	Reading file 'pyart/retrieve/_echo_steiner.pyf' (format:free)
Post-processing...
	Block: _echo_steiner
			Block: convective_radius
			Block: peakedness
			Block: classify
Post-processing (stage 2)...
Building modules...
	Building module "_echo_steiner"...
		Constructing wrapper function "convective_radius"...
		  conv_rad = convective_radius(ze_bkg,area_relation)
		Constructing wrapper function "peakedness"...
		  peak = peakedness(ze_bkg,peak_relation)
		Constructing wrapper function "classify"...
		  sclass = classify(ze,x,y,z,dx,dy,intense,bkg_rad,work_level,area_relation,peak_relation,use_intense,fill_value,[nx,ny,nz])
	Wrote C/API module "_echo_steiner" to file "build/src.linux-i686-2.7/pyart/retrieve/_echo_steinermodule.c"
  adding 'build/src.linux-i686-2.7/fortranobject.c' to sources.
  adding 'build/src.linux-i686-2.7' to include_dirs.
copying /usr/lib/python2.7/dist-packages/numpy/f2py/src/fortranobject.c -> build/src.linux-i686-2.7
[... 617 lines cut ...]
                 from build/src.linux-i686-2.7/fortranobject.h:13,
                 from build/src.linux-i686-2.7/fortranobject.c:2:
/usr/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^
compiling Fortran sources
Fortran f77 compiler: /usr/bin/gfortran -Wall -ffixed-form -fno-second-underscore -fPIC -O3 -funroll-loops
Fortran f90 compiler: /usr/bin/gfortran -Wall -fno-second-underscore -fPIC -O3 -funroll-loops
Fortran fix compiler: /usr/bin/gfortran -Wall -ffixed-form -fno-second-underscore -Wall -fno-second-underscore -fPIC -O3 -funroll-loops
creating build/temp.linux-i686-2.7/pyart/retrieve
creating build/temp.linux-i686-2.7/pyart/retrieve/src
compile options: '-Ibuild/src.linux-i686-2.7 -I/usr/lib/python2.7/dist-packages/numpy/core/include -I/usr/include/python2.7 -c'
gfortran:f90: pyart/retrieve/src/echo_steiner.f90
/usr/bin/gfortran -Wall -Wall -shared build/temp.linux-i686-2.7/build/src.linux-i686-2.7/pyart/retrieve/_echo_steinermodule.o build/temp.linux-i686-2.7/build/src.linux-i686-2.7/fortranobject.o build/temp.linux-i686-2.7/pyart/retrieve/src/echo_steiner.o -lgfortran -o build/lib.linux-i686-2.7/pyart/retrieve/_echo_steiner.so
running build_scripts
creating build/scripts.linux-i686-2.7
copying and adjusting scripts/anytocfradial -> build/scripts.linux-i686-2.7
copying and adjusting scripts/radar_plot -> build/scripts.linux-i686-2.7
copying and adjusting scripts/check_cfradial -> build/scripts.linux-i686-2.7
copying and adjusting scripts/radar_info -> build/scripts.linux-i686-2.7
changing mode of build/scripts.linux-i686-2.7/anytocfradial from 644 to 755
changing mode of build/scripts.linux-i686-2.7/radar_plot from 644 to 755
changing mode of build/scripts.linux-i686-2.7/check_cfradial from 644 to 755
changing mode of build/scripts.linux-i686-2.7/radar_info from 644 to 755
```

jjhelmus, 2015-03-02:
Strange, Anaconda should be building with compile options which include files in `/home/raul/miniconda/lib/` not `/usr/lib/`.  I have to say I'm not sure what is going on here, sorry.