ARM-DOE/pyart #248: Remove _parse_fig_ax() and similar methods from RadarDisplay, make them independent & public

tjlang, 2015-03-03:
I find myself often reproducing outside of Py-ART the functionality of _parse_fig_ax() and similar useful methods in the RadarDisplay family of classes. Since these methods do not rely on any class attributes, it is not necessary for them to be part of those classes. If they were pulled out of these specialized classes, and made independent and public, then one could just import them into other codebases (whether Py-ART-related or not) and not have to actually code them up separately (e.g., from pyart.graph import parse_fig_ax).

jjhelmus, 2015-03-03:
@tjlang Great suggesting.  I'll label this and look into implementing it in the near future.

tjlang, 2015-08-04:
Not sure I care enough about that. It's enough that I can import them if I have Py-ART installed ...

jjhelmus, 2015-08-04:
Keeping the function in the `pyart.graph.common` namespace sounds like a plan for now, we can always revisit this in the future if needed.  Closing this issue as parse_ax_fig is now available.

closed 2015-08-04 by jjhelmus