ARM-DOE/pyart #257: Starting wradlib bridge, how to do integration

scollis, 2015-03-24:
I am working on the first use of wradlib  in Py-ART. We need to:
A) Have the bridge components in the right places (ie io, correct etc..)
B) Allow conditional import as wradlib will be an optional dependency..
The first thing we will start with will be texture calculations as per:
localhost:8888/notebooks/projects/openvap/development/cmac2/notebooks/1 exploration of X and C SAPR data and texture.ipynb

Proposal:
create a new subdirectory pyart/bridge
create file pyart/bridge/wradlib_bridge.py

allow each __init__.py in the various modules to do a conditional import from pyart/bridge/wradlib_bridge.py to populate the namespaces.

Thoughts?

jjhelmus, 2015-03-24:
Concerning the location of the modules and populating the pyart namespace with the various wradlib wrapped function I would prefer to keep all of the bridge and wrapping code in the bridge directory.  A function can then be defined in the bridge `__init__.py` file which populates the pyart.correct, pyart.retrieve, etc namespaces with the wrapped functions.  I made an [example module](https://github.com/jjhelmus/python_dynamic_namespace_test) which does just this.  The [`populate_namespace_with_extra` function](https://github.com/jjhelmus/python_dynamic_namespace_test/blob/master/sayhello/extra/__init__.py#L1) inserts the function defined in the extra directory into the packages namespaces.

scollis, 2015-03-27:
Thanks @jjhelmus I have followed you instructions in PR #259 and testing shows this is a good way.

closed 2015-03-27 by scollis