ARM-DOE/pyart #268: Occasional Py-ART read error on Sigmet files

tjlang, 2015-03-31:
Py-ART occasionally raises a ValueError when ingesting certain Sigmet-format NPOL files. I've placed a number of failure examples on temporary FTP here:
ftp://ftp.nsstc.org/outgoing/tlang/pyart_errors/

Most NPOL files are read fine, maybe 1 in 100 fails. RSL_in_IDL reads these problem files with no errors. This is on a Linux workstation. My Py-ART installation is within a week or so of the current DOE master branch.

Example of the error is below (pyblock.py is my program calling pyart.io.read):

radar file: np1130520022537.RAWGFG6 & sounding: default_sounding.txt
Traceback (most recent call last):
  File "ifloods_blockage_step01.py", line 16, in <module>
    image_dir='/raid1/tlang/ifloods/')
  File "/raid1/tlang/anaconda/envs/py27/lib/python2.7/site-packages/pyblock-1.0-py2.7.egg/pyblock/pyblock.py", line 405, in __init__
    self.process_multiple_volumes()
  File "/raid1/tlang/anaconda/envs/py27/lib/python2.7/site-packages/pyblock-1.0-py2.7.egg/pyblock/pyblock.py", line 421, in process_multiple_volumes
    self.kwargs['sounding'] = self.sounding_list[i]
  File "/raid1/tlang/anaconda/envs/py27/lib/python2.7/site-packages/pyblock-1.0-py2.7.egg/pyblock/pyblock.py", line 128, in __init__
    radar = pyart.io.read(filename)
  File "/raid1/tlang/anaconda/envs/py27/lib/python2.7/site-packages/pyart/io/auto_read.py", line 124, in read
    return read_sigmet(filename, **kwargs)
  File "/raid1/tlang/anaconda/envs/py27/lib/python2.7/site-packages/pyart/io/sigmet.py", line 136, in read_sigmet
    sigmet_data, sigmet_metadata = sigmetfile.read_data(full_xhdr=full_xhdr)
  File "pyart/io/_sigmetfile.pyx", line 180, in pyart.io._sigmetfile.SigmetFile.read_data (pyart/io/_sigmetfile.c:4595)
  File "pyart/io/_sigmetfile.pyx", line 259, in pyart.io._sigmetfile.SigmetFile._get_sweep (pyart/io/_sigmetfile.c:5639)
  File "pyart/io/_sigmetfile.pyx", line 804, in pyart.io._sigmetfile._unpack_ingest_data_headers (pyart/io/_sigmetfile.c:11601)
  File "pyart/io/_sigmetfile.pyx", line 818, in pyart.io._sigmetfile._unpack_ingest_data_header (pyart/io/_sigmetfile.c:11910)
ValueError

scollis, 2015-03-31:
By occasionally is it seemingly random? ie is there anything different 
about these scans (start of change of VCP?)

jjhelmus, 2015-03-31:
@tjlang Can you send an example of a file which causes this issue, I should be able to debug the problem with an example and put together a fix.

The ValueError is being raised because the structure identifier in the ingest data header does not match the expected value of 24 which indicated an Ingest_data_header, [line 817 in _sigmetfile.pyx](https://github.com/ARM-DOE/pyart/blob/master/pyart/io/_sigmetfile.pyx#L817).  Basically this means that the code is no longer reading data that it expects to be reading so it errors outs.  Most likely something changed earlier in the file that the code failed to take into account, with an example I should be able to figure out what.

tjlang, 2015-03-31:
Seems random, no major shifts in scan type - these should be all 2-tilt rain-map PPI surveillance sweeps. Using IDL I can confirm the files contain real data with no obvious problems. The ValueError appears related to the data header as @jjhelmus notes. As I noted in my original post, I've placed a number of failure examples on temporary FTP here:
ftp://ftp.nsstc.org/outgoing/tlang/pyart_errors/

jjhelmus, 2015-03-31:
Do you get a two tilt volume when reading with IDL?  The C TRMM RSL library sees two sweeps in the file but detects errors in the second sweep and ignores that data:

```
$ /usr/local/trmm/bin/any_to_uf data/np1130520022537.RAWGFG6 test.ufopen file: data/np1130520022537.RAWGFG6
Read 6144 bytes for rec1.
ID = 27
Read 6144 bytes for rec2.
Expecting 360 rays in each sweep.
nparams = 13, nsweeps = 2
...
     nsig_sweep[12], data_type = 66, rays(expected) = 360, nrays(actual) = 300
nsig_read_ray: chunk return code = -1.
nsig_read_ray: chunk return code = -1.
nsig_read_ray: chunk return code = -1.
...
$ radar_info test.uf
...
nsweeps: 1
...
```

jjhelmus, 2015-03-31:
In the file `np1130520022537.RAWGFG6` there are two records which hold the data for the second sweep (records 1054 and 1055).  Both contain a valid 12-byte raw_prod_bhdr structure followed by 0x10 repeated to fill the record which is invalid.  The other file I examined, `np1130520024112.RAWGFJ2`, has a similar layout.  It should be possible to have Py-ART issue a warning when this is detected and return the first sweep, which is in line with what **pyart.ioread_rsl** gives. 

@tjlang Would returning only the first sweep be the expected behavior for these files?  Is that what IDL does?

jjhelmus, 2015-03-31:
A fix which allows for partial reading of these files available in PR #269.  On the example files the first sweep is read and a UserWarning issues.

tjlang, 2015-03-31:
Sorry for the delay, I agree that the second tilt appears corrupted. RSL_in_IDL is able to ingest the file but just populates the second tilt with zeros (e.g., for reflectivity). So a fix that populates the good sweep and ignores the rest is acceptable.

jjhelmus, 2015-03-31:
Exactly what PR #269 does, merging once the CI tests pass.

```
$ python -c "import pyart; r = pyart.io.read_sigmet('data/np1130520022537.RAWGFG6'); print r.nsweeps"
/home/jhelmus/dev/pyart/pyart/io/sigmet.py:136: UserWarning: File truncated or corrupt, 1 of 2 sweeps read
  sigmet_data, sigmet_metadata = sigmetfile.read_data(full_xhdr=full_xhdr)
1
```

Thanks for catching this one!

PR #269 (2015-03-31): FIX: Partial reading of corrupt sigmet files.
When a sigmet/IRIS file abruptly ends with an invalid ingest_data_header,
all sweeps read up to that point are returned and a warning is issues.

This fixes #268
commit b838cd7f (2015-03-31): FIX: Partial reading of corrupt sigmet files.

When a sigmet/IRIS file abruptly ends with an invalid ingest_data_header,
all sweeps read up to that point are returned and a warning is issues.

This fixes #268
closed 2015-03-31 by jjhelmus