ARM-DOE/pyart #270: ARM ingest data

kirknorth, 2015-03-31:
Just to clarify, are we currently unable to read ARM ingest data? For example, this [KAZR](https://www.dropbox.com/s/cu7h8hjfkk9xgzf/sgpkazrgeC1.a1.20110511.000000.cdf?dl=0) file? @jjhelmus I know you've tested the new dealiasing routines on 915-MHz RWP data and I wanted to do something similar with KAZR data.

jjhelmus, 2015-03-31:
Py-ART should be able to read the a1 level files for the scanning ARM radars, SACRs and XSAPR but is currently unable to properly read data from the zenith pointing instruments, RWP, KAZR, MMCR files.  Although these are in netCDF format they are not Cf/Radial netCDF.  If you wanted to write a reader for a particular instrument or attempt to make a general purpose ARM zenith pointing read including it in **pyart.aux_io** or even **pyart.io** would be reasonable.

For the RWP dealiasing I wrote a one off script which created an empty Radar object using `pyart.testing.make_empty_ppi_radar` and added the velocity data.  The script had a number of hard coded values to get things working but it got the job done.  I can share the full script if you would like but I don't think it would be of much use.  

The key snippet which I would only use to get something working quickly is:

``` Python
radar = pyart.testing.make_empty_ppi_radar(75, 800, 1)
radar.fields['velocity'] = {'data': vel}
radar.fields['snr'] = {'data': snr}

nyquist_vel = dset.variables['vband'][0] / 2.
gatefilter = pyart.correct.GateFilter(radar)
gatefilter.exclude_below('snr', -10)
gatefilter.exclude_invalid('snr')

corr_vel = pyart.correct.dealias_region_based(
    radar, vel_field='velocity', keep_original=True,
    gatefilter=gatefilter, skip_along_ray=10, rays_wrap_around=False,
    nyquist_vel=nyquist_vel)
radar.add_field('corrected_mean_doppler_velocity', corr_vel)
```

scollis, 2015-03-31:
I will also add one of the things that was stopping us ingesting the ARM 
KAZR format is that it will eventually move to CF-Radial.

kirknorth, 2015-03-31:
Cool, thanks for the quick response guys! @jjhelmus I'll look into developing something for this format that could eventually be included in **aux_io**. I think the auxiliary module is more relevant since, as @scollis points out, these formats will eventually be deprecated.

closed 2022-03-09 by zssherman