ARM-DOE/pyart #281: Dealiasing VPT data

kirknorth, 2015-04-16:
@jjhelmus and @scollis from a past email I mentioned that the phase unwrap algorithm had difficulties with the insect and shallow cumulus layer, but I now have firm evidence that both the new algorithms have issues with this layer. Here's a [link](http://1drv.ms/1OjK7L4) to a full days worth of KAZR data that I have processed using `dealias_region_based`, where I have used the default settings. This shallow layer is incorrectly dealiased for about 20 hours, where at approximately 2033 UTC and afterwards the results improve. I think the cloud *feature* responsible for this sudden change is obvious and is almost certainly the reason for the poor results for times before 2033 UTC.

scollis, 2015-04-16:
hey @kirknorth Just to be sure is it this: 
https://onedrive.live.com/?cid=61ABE98B46A2C558&id=61abe98b46a2c558!2128&parId=61abe98b46a2c558!2130&v=3&authkey=!AAveyH6OdW52Kfs

kirknorth, 2015-04-16:
@scollis that's hour 22, you should have access to the full day.

scollis, 2015-04-17:
Very interesting.

This is where we may need to bring in a little more prior information. 
It identifies it as a segment but ascribes the wrong fold

What gatefilter are you using?

kirknorth, 2015-04-17:
For the insect layer spanning the first 20 hours or so I wouldn't say it prescribes the wrong fold, it shouldn't be prescribing anything for this layer since it's not folded and the initial velocities are good. The gate filter is defined by the lower left panel, i.e. the significant detection mask, which is essentially a combination of the noise floor mask and the velocity coherency mask.

jjhelmus, 2015-04-17:
@kirknorth How large of time period were the dealiasing routines run on in a single pass?  Both algorithms make the assumption that the velocities in the sweeps or chunks of data they operate on should be smoothly varying.  In the data linked it looks like a large shear event at approximate 2030 UTC is forcing all velocities prior to that time to have an extra nyquist added.  If this was excluded from the analysis my guess it that the corrections would be improved.  Not exactly a solution but an explanation...

Next week I'm hoping to have some time to add some debugging and logging code to the region dealiasing algorithm so I can get a better handle on what is going on during cases like this.  I have some ideas but I'd like to verify them before speculating on the cause and suggesting solutions.  

One simple idea to try  is to rerun the algorithm with a higher value for the **interval_splits** parameter.  Setting that parameter to 4 or 5 will further segment the initial velocity.  The algorithm will take longer to complete but may give improved results.

kirknorth, 2015-04-17:
@jjhelmus the time period was simply the full record, i.e., 24 hours, which, as you point out, is likely the underlying issue that I did not consider. I will test some larger values of **interval_splits** and report back.

kirknorth, 2015-04-21:
Increasing **interval_splits** to 5 (instead of the default 3) did improve the results within the insect layer before the cloud observed near 2030 UTC.

[image]

However looks like the `dealias_region_based` algorithm still has issues with the cloud itself. Collocated Doppler lidar data indicates a cloud base near 1.5 km near this time. However, I find the DL data raises more questions than answers. Basically I don't see a prominent updraft in the DL data between 2030 and 2035 UTC. @scollis can you comment on this? What am I missing? Is it simply that these instruments aren't as collocated as I'm assuming, because there does appear to be prominent upwards motion between 2025 and 2030 UTC (too bad there's a change of modes here), as well as between 2035 and 2040 UTC.

[image]

scollis, 2015-04-21:
> sgpkazrgec1 a1 20110523 000002
> 20110523 200000

Kirk, What day?
Look at the sounding and the LCL. and also the TSI..
That cloud is super low. Separation between DL *NOW* and KAZR is ~10m.. 
so we expect reasonable correlation.

use http://plot.dmf.arm.gov/plotbrowser/ to deep dive

kirknorth, 2015-04-21:
This is a MC3E case from 2011-05-23.

scollis, 2015-04-21:
Sounding before (20Z)
http://plot.dmf.arm.gov/PLOTS/SGP/sgpsondewnpn/20110523/sgpsondewnpnC1.b1.skewt.201900.20110523.png
Vceil (~200m away)
http://plot.dmf.arm.gov/PLOTS/SGP/sgpvceil25k/20110523/sgpvceil25kC1.b1.backscatter.20110523.png

Ok here is where things get super confusing!
TSI shows lots of clouds.. 
http://plot.dmf.arm.gov/PLOTS/SGP/sgptsiskycover/20110523/sgptsiskycoverC1.b1.tsimovie.20110523.mp4
I know it is hard to tell but to my thoughts these look low..

scollis, 2015-04-21:
BTW.. I am assuming the missing data in the DL is due to a PPI scan..

kirknorth, 2015-04-21:
Well the DL intensity and backscatter data for this hour actually indicates a far amount of clouds too, some (most) of which are not obvious in the KAZR data, likely due to sensitivity. For example, the gap in clouds as seen in the DL data between 2040 and 2050 UTC is also observed in the TSI during this time.

kirknorth, 2015-04-21:
The collocated ceilometer tells a similar story.

[image]

kirknorth, 2015-04-22:
I'm now convinced we shouldn't be seeing an updraft in the DL data. Instead the KAZR is most likely detecting a few large drops falling towards it, meaning that the "rain shaft" (not actually raining) near 2030 UTC gets prescribed the wrong fold by `dealias_region_based`, i.e. it should be negative velocities implying motion towards the KAZR.

scollis, 2015-04-23:
Hey all Virendra suggested the rainshaft may be sheared and be from a cloud out of view of the radar. Note the dealiaser had nothing to go on. No apriori telling it what fold.. This would need to be built in post processing

kirknorth, 2015-04-23:
@scollis there is a region (segment) of downwards (negative) velocities in the raw data associated with this feature, which would suggest this *could potentially* be corrected properly, but maybe this is pushing the limits of automatic methods, in particular `dealias_region_based`? @jjhelmus I look forward to your comments.

scollis, 2015-04-23:
Yes.. But the key here is that the region based approach was designed, 
first, to work without any other knowledge. As the regions are stored 
future data could be used.. But as you said @jjhelmus 
<https://github.com/jjhelmus> may be able to suggest a set up..

BTW I agree that the velocities should, indeed, be towards the data.. 
Makes me want a distrometer at the KAZR site. Diameter fallspeed could 
end up being a constraint for this algorithm and ARSCL2 in the future.

kirknorth, 2015-04-23:
But yes, as Vinny suggested, if this is a case of a sheared part of one cloud coming into view of the radar and merging and/or embedding with other clouds in view than the velocity solution becomes ambiguous here.

jjhelmus, 2015-04-24:
From a quick look at the raw and corrected velocities I don't think the region based algorithm is going to be able to unfold the upward component on the right of the sheared cloud into a downward velocity since the upward region is larger are more organized than the downward region on the right.  I'll have to run through the case through the algorithm with some verbose information turned on to tell for sure what is going on but I'm not hopeful.

kirknorth, 2015-04-24:
To inject a positive note, the results also improved for hour 11 when increasing **interval_splits** to 5. Here were the [original results](http://1drv.ms/1z1G4S8).

[image]

scollis, 2017-07-13:
I think this conversation is done.. closing

closed 2017-07-13 by scollis