ARM-DOE/pyart #298: I got a problem when I install pyart

qwe14789cn, 2015-05-19:
I'm working @ubuntu 14.10 + Anaconda.
And I setup as your guide:

conda update conda
conda install basemap
conda install -c jjhelmus pyart

but when i work in ipython
>>>  import pyart

I got error like this:

import pyart
ERROR: An unexpected error occurred while tokenizing input
The following traceback may be corrupted or invalid
The error message is: ('EOF in multi-line string', (1, 2))

---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-706b47c96bb9> in <module>()
----> 1 import pyart

/home/qwe14789cn/pyart-master/pyart/__init__.py in <module>()
     18 
     19     # print out helpful message if build fails or importing from source tree
---> 20     from . import __check_build
     21 
     22     # versioning

/home/qwe14789cn/pyart-master/pyart/__check_build/__init__.py in <module>()
     44     from ._check_build import check_build
     45 except ImportError as e:
---> 46     raise_build_error(e)

/home/qwe14789cn/pyart-master/pyart/__check_build/__init__.py in raise_build_error(e)
     39 to build the package before using it: run `python setup.py install` in the
     40 source directory.
---> 41 %s""" % (e, local_dir, ''.join(dir_content).strip(), msg))
     42 
     43 try:

ImportError: No module named _check_build
___________________________________________________________________________
Contents of pyart/__check_build:
_check_build.c            README                    setup.py
_check_build.pyx          setup.pyc                 __init__.py
__init__.pyc
___________________________________________________________________________
It seems that Py-ART has not been built correctly.

If you have installed Py-ART from source, please do not forget
to build the package before using it: run `python setup.py install` in the
source directory.

It appears that you are importing a local pyart source tree. For this, you
need to have an inplace install. Maybe you are in the source directory and
you need to try from another location.

I also install @windows7 .it didn't happen.

could you tell me how to fix it?
Thanks

qwe14789cn, 2015-05-19:
I solved it.
ubuntu14.10 + Anaconda
1.
gedit ~/.bashrc
add code:
export LC_ALL=en_US.UTF-8
export LANG=en_US.UTF-8

source ~/.bashrc

then do not use 'conda install -c jjhelmus pyart'

use 
git clone https://github.com/ARM-DOE/pyart.git
python setup.py install --user

import pyart   
succeed!

jjhelmus, 2015-05-20:
@qwe14789cn Has this issue been solved or do you need additional help?  If this is working can this issue be closed?

qwe14789cn, 2015-05-20:
This problem could be closed.But I suggest use
"Python setup.py install --user"
Instead of
"conda install -c jjhelmus pyart"
Use conda install pyart may bring some bug.
The right order should be

Conda update conda
Conda install basemap
Python pyart install --user

This bug only @ubuntu+anaconda
Windows7+anaconda is ok

jjhelmus, 2015-05-20:
The INSTALL.rst and README.rst files do not mention conda as a method to install Py-ART, only the Wiki recommends using conda, which I will remove for Linux.  The version of Py-ART on my binstar channel is quite old and will not work on all Linux distributions or versions due to difference in the glib. I have some ideas on how to improve this situation and build more universal Linux conda packages, I'll tests these out in the next release. In addition, if conda is used to install Py-ART on any platform, the install will only reliably work with the Python interpreter and libraries provided by Continuum in either Anaconda or Miniconda.

closed 2015-05-20 by jjhelmus