ARM-DOE/pyart #299: nbins is an integer in sigmet reader

WeatherGod, 2015-05-19:
Working off of master and reading in some IRIS data, I get the following traceback:

```
>>> rad = pyart.io.read('data/201211221820~~CONVOL:URP:WKR:RADAR:IRIS')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/broot/scratch/miniconda/lib/python2.7/site-packages/pyart/io/auto_read.py", line 133, in read
    return read_sigmet(filename, **kwargs)
  File "/home/broot/scratch/miniconda/lib/python2.7/site-packages/pyart/io/sigmet.py", line 136, in read_sigmet
    sigmet_data, sigmet_metadata = sigmetfile.read_data(full_xhdr=full_xhdr)
  File "pyart/io/_sigmetfile.pyx", line 180, in pyart.io._sigmetfile.SigmetFile.read_data (pyart/io/_sigmetfile.c:4596)
  File "pyart/io/_sigmetfile.pyx", line 298, in pyart.io._sigmetfile.SigmetFile._get_sweep (pyart/io/_sigmetfile.c:6285)
  File "pyart/io/_sigmetfile.pyx", line 752, in pyart.io._sigmetfile.convert_sigmet_data (pyart/io/_sigmetfile.c:10625)
TypeError: Cannot convert int to numpy.ndarray
```
Looking at line 752 in _sigmetfile.pyx, I see that it is passing in a view and an integer `nbins` to the function `_mask_gates_not_collected()`, which expects two ndarrays. The function this is called in already takes an argument `nbins`, and this one if-branch is the only place where `nbins` gets re-assigned (at line 677). I think we only need to change `nbins` to `nbin` at line 677 and 678, but I haven't been able to get conda-build to work properly.

WeatherGod, 2015-05-19:
Ah, it isn't conda-build that is the issue, the pyart/io/_sigmetfile.c source isn't getting redone...

WeatherGod, 2015-05-19:
I see the `rebuild_sigmetfile.sh`, and running it does remake the .c file. So I did that after doing a `git clean -fxd`, and then did a `python setup.py install` in a conda environment. Importing now gets me some strange errors, so I suspect I did something wrong in the build process.

WeatherGod, 2015-05-19:
Ok, I think I got the build process figured out (I had originally cython _sigmetfile.c with an older version of cython, which caused all sorts of weirdness at import). Using cython 0.22, and redoing the .c file and all that, the file reads without errors!

WeatherGod, 2015-05-20:
So, it would seem that the following diff for the pyx file is needed:
```
diff --git a/pyart/io/_sigmetfile.pyx b/pyart/io/_sigmetfile.pyx
index 0283165..67d3f88 100644
--- a/pyart/io/_sigmetfile.pyx
+++ b/pyart/io/_sigmetfile.pyx
@@ -674,8 +674,8 @@ def convert_sigmet_data(data_type, data, nbins):
     elif data_type_name[-1] != 2:
         # make a view of left half of the data as uint8,
         # this is the actual ray data collected, the right half is blank.
-        nrays, nbins = data.shape
-        ndata = data.view('(2,) uint8').reshape(nrays, -1)[:, :nbins]
+        nrays, nbin = data.shape
+        ndata = data.view('(2,) uint8').reshape(nrays, -1)[:, :nbin]
 
         if data_type_name in like_dbt:
             # DB_DBT, 1, Total Power (1 byte)
```
since I didn't use the exact same version of cython, the diff in the .c file is much more extensive. Do you want me to make a PR with just this change?

scollis, 2015-05-20:
Thanks Ben,
We will get to this soon.

jjhelmus, 2015-05-20:
Yes nbins should not be redefined on that line.  If you want to make a PR with the change to the .pyx and .c file that would be great.  Adding a unit test to catch this bug would be even better.

jjhelmus, 2015-05-20:
Also thanks for catching this @WeatherGod.

WeatherGod, 2015-05-20:
No problem. I am putting together the commit now using cython 0.21.2.

Question: for the unit test, I could include an example 1byte format sigmet file, but I see that this project typcially just generates example data. I don't know anything about sigmet data formats to know what makes it a 1byte format, so would it still be ok to include the example that I have?

WeatherGod, 2015-05-20:
I don't see any sort of suite of tests for reading the example files. Does one need to be created? I see tests for configuration handling, and a smoketest for debug output.

jjhelmus, 2015-05-20:
The tests for the **_sigmetfile** module are combined with those for the **sigmet** module in [*test_sigmet.py*](https://github.com/ARM-DOE/pyart/blob/master/pyart/io/tests/test_sigmet.py) in the *pyart/io/tests* directory.  I don't think that full 1byte format sigmet file must be tested, feeding data to the `convert_sigmet` function should be sufficient, for example:

``` Python
import pyart
import numpy as np
data_type = 1   # Sigmet type for for a DBT field 
data = np.ones((2, 2), dtype=np.int16) * 257
nbins = np.ones((2,), dtype=np.int16)
result = pyart.io._sigmetfile.convert_sigmet_data(data_type, data, nbins)
```

I believe this should return a 2x2 array containing -31.5 but currently raises a TypeError.

WeatherGod, 2015-05-20:
Indeed, running the above code snippet on my patched install works:

```
>>> result
masked_array(data =
 [[-31.5 --]
 [-31.5 --]],
             mask =
 [[False  True]
 [False  True]],
       fill_value = -9999.0)
```
I take it we are expecting half the data to be masked out, though? Like I
said, I am totally clueless on sigmet/iris data formats. If that is all
good, then I can easily write up a test for that. Thanks for pointing me in
the right direction. I am too used to keeping all my tests in one place.

jjhelmus, 2015-05-20:
Oops add a * 2 to the nbins line `nbins = np.ones((2,), dtype=np.int16) * 2` and then you should get all -31.5 and no masking.  But yes that is the expected values for that type of sigmet/iris data.

PR #305 (2015-05-21): Fixed issue reading sigmet 1byte datatype files. Added test.
Closes #299
closed 2015-05-21 by jjhelmus