ARM-DOE/pyart #300: Trouble in write_cfradial

seberry2, 2015-05-20:
Hello,
I am trying to write a netCDF file, however I keep getting the following error: 

    pyart.io.write_cfradial(filesavename, radar)
  File "C:\Anaconda\lib\site-packages\pyart\io\cfradial.py", line 527, in write_cfradial
    _create_ncvar(start_dic, dataset, 'time_coverage_start', time_dim)
  File "C:\Anaconda\lib\site-packages\pyart\io\cfradial.py", line 661, in _create_ncvar
    ncvar[..., :data.shape[-1]] = data[:]
  File "netCDF4.pyx", line 3319, in netCDF4.Variable.__setitem__ (netCDF4.c:46671)
  File "C:\Anaconda\lib\site-packages\netCDF4_utils.py", line 179, in _StartCountStride
    dimname = dimensions[i]
IndexError: tuple index out of range

The file actually appears and sees to have all the correct information in it. I tired reading in a file and then immediately trying to write it out, but I obtained the same error as above. 

Any ideas?

Thanks!

jjhelmus, 2015-05-20:
@seberry2 Can you provide some additional details about your setup.  Specifically the version of Py-ART  you are using as well as the netCDF4 version would be extremely helpful.  Executing `python -c "import pyart; pyart._debug_info()"` on the command line should provide enough information about your setup.  

My suspicion is that this is an issue with netCDF4 1.1.6+ which was fixed in 7d9e1a89cf8c4d07

seberry2, 2015-05-20:
I am using Py-ART version 1.3.0 and netCDF4 version 1.1.5.

jjhelmus, 2015-05-20:
My best guess is still that this error was fixed in 7d9e1a8.  Can you either update Py-ART to the latest version on GitHub or change to a version of netCDF4 not effected by this ellipsis bug, version 1.1.1 or 1.1.7+ should work, and see if this solves the issue?

Alternatively, if you can send me the file and a script which produces the error, I can try to repeat the process on my machine and nail this down.

seberry2, 2015-05-20:
Installing netCDF4 version 1.1.1 solved the problem. Thanks!

jjhelmus, 2015-05-20:
Your welcome, once Py-ART 1.4.0 is released you should be able to upgrade netCDF4 to 1.1.5 or later.  Closing issue.

closed 2015-05-20 by jjhelmus