ARM-DOE/pyart #318: _gate_to_grid_map module does not compile on Windows

jjhelmus, 2015-06-09:
The _gate_to_grid_map module does not compile on Windows as the Microsoft Visual Studio C++ compiler used to compile Python 2.7 modules does not implement the C99 isfinite function.  The current work around is to disable the check for finite field values.  

In addition Windows does not separate the math library from the standard C library so the "m" library is not need nor can be linked to the module.

jjhelmus, 2015-06-09:
The second portion of this bug is fixed by PR #319

gamaanderson, 2015-06-09:
Just for information, I have found the same problem (the first one) while compiling in Linux, but I cann't say now what configuration in was. The problem, I believe, eventually solved itself.

jjhelmus, 2015-06-09:
Linking against a libc/libm library on Linux which does not implement the C99 isfinite function would cause such an error but you would need to use a very old version of the GNU C library or a non-standard C library for that.  I think long term all filtering of gates for gridding should be handles by the gatefilter object.  Then isfinite would not be needed in the _gate_to_grid_map module.

jjhelmus, 2015-08-24:
I was able to build Py-ART successfully on Windows with PR #360 merged. Closing this issue as it has been resolved.

closed 2015-08-24 by jjhelmus