ARM-DOE/pyart #321: Configuring Names

gamaanderson, 2015-06-19:
Would it be possible to the user to change the default py-art name ('reflectivity' etc.) while **installing**. I know all functions have options to overwrite the standard, but we have here in SIMEPAR a really (internaly)-uniform standard, but as it is different than py-art I must always correct it. ARTview is also having some difficulties because of that.

jjhelmus, 2015-06-19:
The field names and other configuration details are set in the Py-ART configuration file.  I think we can include an option during the install to copy a custom configuration file into the pyart directory and adjust the configuration loading logic to use this configuration file if available.  I would still want the `default_config.py` file available for testing and such.  

I should add that currently a custom configuration file will be loaded during import if the  **PYART_CONFIG** environment variable is set and points to a valid configuration file. This is the suggested method for using a different configuration for the time being.  This is detailed in the documentation of [load_config](http://arm-doe.github.io/pyart-docs-travis/user_reference/generated/pyart.load_config.html#pyart.load_config)

zssherman, 2020-03-26:
Documentation was added. Closing.

closed 2020-03-26 by zssherman