ARM-DOE/pyart #325: Error on "native" sigmet read

deeplycloudy, 2015-06-25:
I have a Sigmet file which crashes pyart.io.read_sigmet. A call to pyart.io.read_rsl works fine. See the interactive session below; something seems suspicious about the call to create the masked array. I'm running master from a half hour ago. 

A secondary issue is that this file is reported as a PPI by RSL, while it's actually 10 RHIs. I'd like to see how the scan mode is encoded, which is why I was hoping to get at the lower-level header data exposed by the native Sigmet support.

File is here:
https://dl.dropboxusercontent.com/u/5587496/RAW_NA_000_025_20130606030822.gz

```
In [3]: ka1_sigmet=pyart.io.read_sigmet('/data/20130606/Ka1uf/RAW_NA_000_025_20130606030822.gz')
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-3-9f23184683af> in <module>()
----> 1 ka1_sigmet=pyart.io.read_sigmet('/data/20130606/Ka1uf/RAW_NA_000_025_20130606030822.gz')

/Users/ebruning/anaconda/lib/python2.7/site-packages/pyart/io/sigmet.pyc in read_sigmet(filename, field_names, additional_metadata, file_field_names, exclude_fields, time_ordered, full_xhdr, noaa_hh_hdr, debug, ignore_xhdr, ignore_sweep_start_ms, **kwargs)
    138 
    139     # read data
--> 140     sigmet_data, sigmet_metadata = sigmetfile.read_data(full_xhdr=full_xhdr)
    141     first_data_type = sigmetfile.data_type_names[0]
    142     if first_data_type == 'XHDR':   # don't use XHDR as the first data type

pyart/io/_sigmetfile.pyx in pyart.io._sigmetfile.SigmetFile.read_data (pyart/io/_sigmetfile.c:3961)()

/Users/ebruning/anaconda/lib/python2.7/site-packages/numpy/ma/core.pyc in __call__(self, a, *args, **params)
   7312             _extras[p] = params.pop(p)
   7313         # Get the result
-> 7314         result = self._func.__call__(a, *args, **params).view(MaskedArray)
   7315         if "fill_value" in common_params:
   7316             result.fill_value = _extras.get("fill_value", None)

ValueError: negative dimensions are not allowed

In [4]: 

In [4]: %debug
> /Users/ebruning/anaconda/lib/python2.7/site-packages/numpy/ma/core.py(7314)__call__()
   7313         # Get the result
-> 7314         result = self._func.__call__(a, *args, **params).view(MaskedArray)
   7315         if "fill_value" in common_params:

ipdb> print a
(-31868, 20858, 772334215)
ipdb> print args
()
ipdb> print params
{'dtype': 'float32'}
```

jjhelmus, 2015-06-25:
Looking into this, should have so ideas in a bit

jjhelmus, 2015-06-25:
@deeplycloudy The file (RAW_NA_000_025_20130606030822.gz) you are looking at is a gzipped UF file so `read_sigmet` will not read the file.  Currently the only method for reading UF files in Py-ART is to use the `read_rsl` function.   The UF header does indicate that the generation facility was IRIS8.11 so this data was likely at one time in Sigmet/IRIS format.

When I read the  file after unpacking with `pyart.io.read` or `pyart.io.read_rsl` I get back a 10 sweep RHI radar object.  Strangely, if I read the file a second time in the same session I get back a 10 sweep PPI radar object which makes no sense.  I'll need to investigate to see if this is an issue with Py-ART or RSL.

jjhelmus, 2015-06-25:
The issue with getting different scan modes when reading in the same file twice is an issue in RSL not Py-ART.  For example:

``` C 
#include "rsl.h"

void main(int argc, char **argv) {

    Radar *radar;
    
    RSL_print_version();

    // read the file, print the scan_mode, 0: PPI, 1: RHI
    radar = RSL_uf_to_radar("RAW_NA_000_025_20130606030822");
    printf("scan_mode: %i\n", radar->h.scan_mode);
    RSL_free_radar(radar);
    
    // re-read the file and print scan mode
    radar = RSL_uf_to_radar("RAW_NA_000_025_20130606030822");
    printf("scan_mode: %i\n", radar->h.scan_mode);
}
```

Produces:
```
$ ./a.out 
RSL version v1.47.
scan_mode: 1
scan_mode: 0
```

I'll let Bart Kelley know about this issue and see if there is a fix.

jjhelmus, 2015-06-25:
Marking this as **Wontfix** for the time being as this needs to be addressed in RSL and not Py-ART.  If/when Py-ART has a native Uf reader (#201) then fixing this in RSL will be less important.

deeplycloudy, 2015-06-25:
Ugh, I'm sorry about sending you UF. My actual Sigmet files, once I actually used them, worked great and reported successfully as RHIs. 

Interesting bug in the RSL reader!

jjhelmus, 2015-06-26:
No problem @deeplycloudy.  Bart Kelley provided me with an updated uf_to_radar.c file for RSL which fixes the bug when UF file are read multiple times.  This change will be included in the next release of RSL so I've going to close this issue.

closed 2015-06-26 by jjhelmus