ARM-DOE/pyart #329: Default field colormaps and limits specified in Py-ART configuration file.

jjhelmus, 2015-07-09:
As discussed in [ARTView issue #43](https://github.com/nguy/artview/issues/43) it would be helpful if default colormaps and field limited were specified in the Py-ART configuration file.

gamaanderson, 2015-07-16:
What do you think about [this addition](https://github.com/simepar/pyart/blob/master/pyart/default_config.py#L1101-L1258) to solve this issue?

P.S. I'm not sure about [this line](https://github.com/simepar/pyart/blob/master/pyart/default_config.py#L1110), how should I import pyart, relative or absolute?

jjhelmus, 2015-07-20:
I'd rather have the configuration file contain contain tables of values not functions.  The functions should be placed in the **config.py** module which read parameters from these tables.  This is how the `get_metadata`, `get_fillvalue` and `get_field_name` functions are implemented.  I have a number of projects that need attention this week but this topic is on my to do list.

gamaanderson, 2015-07-20:
The idea was to give the user the opportunity of adding other fields, so those functions should stay in **default_config.py** (only `get_colormap` I agree to put in **config.py**). The difference in this task is that I don't want to recognize some specific field (like reflectivity), but to handle all fields/quantities that may be in a file.

jjhelmus, 2015-07-20:
I do not like the idea of having a multitude of functions which serve only to return a tuple of static data, that situation is better served with a dictionary.  In cases where introspection is needed (Doppler velocity) a function is the only option but I do not think that requires that all the field limits and colormaps should be defined in functions.  Rather I think the `callable`  could be used to check if the particular object in the dictionary is a callable function and then defer to that.  Also I would seperate the colormap from the limits.  Something along the following:

In default_config:

``` Python
DEFAULT_FIELD_COLORMAPS = {
    'reflectivity': 'pyart_NWSRef',
    'velocity': 'pyart_NWSVel', 
    # complete for all fields in DEFAULT_FIELD_NAMES, some will be identical
}

DEFAULT_FIELD_LIMITS = {
    'reflectivty': (-10, 65),
   'velocity':  _velocity_limits
    # complete for all fields in DEFAULT_FIELD_NAMES, some will be identical
}

def _velocity_limits(radar, sweep):
    try:
         nyquist = radar.get_nyquist_vel(sweep)
    except:
         nyquist =  30
   return -nyquist, nyquist
```

In config.py

``` Python
# globals _DEFAULT_FIELD_COLORMAP and _DEFAULT_FIELD_LIMITS defined and read by load_config function

def get_field_colormap(field):
   if field in _DEFAULT_FIELD_COLORMAP:
       return _DEFAULT_FIELD_COLORMAP[field]
   else:
     import matplotlib.cm
     return matplotlib.cm.get_cmap()

def get_field_limits(field, radar, sweep):
    if field in _DEFAULT_FIELD_LIMITS:
      limits = _DEFAULT_FIELD_LIMITS[field]
      if callable(limits):
        limits = limits(radar, sweep)
     return limits
    else:
      return None, None   # or something else?
```

gamaanderson, 2015-07-20:
Ok, that work for me, I will implement and create a PR, however one still need to revise my chooses as I don't have the meteorologic knowledge

jjhelmus, 2015-07-21:
Let me know when you have a PR and I can have a look and help with some of the defaults. @scollis might be able to provide some good limits also.

gamaanderson, 2015-07-22:
I may have agree to fast to your proposal. Two points before I can implement this:

1. the limits function is getting radar and sweep, I would rather name those as dataContainer and tilt, basic in anticipation for receiving grids or other data structures.

2. the DEFAULT* dicts keys are the internal field name, but I've rather giver the user the possibility of adding more fields, for instance "reflectivity_horizontal", "reflectivity_vertical", "right_hand_reflectivity" etc., however once the names used are the internal ones adding others is possible but  in my understanding not "allowed". Also related to that: what do you think of the [PHYSICAL_QUANTITIES](https://github.com/simepar/pyart/blob/master/pyart/default_config.py#L1186) dict? This will avoid the repetitions.

jjhelmus, 2015-07-22:
@gamaanderson  Addressing your two points:

1. Agreed that we should be able to pass a instance of a Grid or Radar object or any future classes. I would prefer the parameter names **container** and **selection** with some documentation on what these typically represent.

2. Having all the internal field names present in the DAFAULT_* dictionaries is a minimal requirement in order that all default field names have a default colormap and limits.  This will also make these dictionaries play nicely with the simple customization section of the file. The user (maybe even the default configuration) could include additional field names mappings although this could cause conflicts if the default field names are changed. Such addition would be fully supported and recommended.  The preference here would be files have default field names and everything works nicely out of the box, but alternative field names would be supported given their prevalence. 

As for using a intermediate PHYSICAL_QUALTITIES dictionary, I'm against it.  I think it added an unnecessary level of abstraction and complication while saving very little in terms of repetition.   To quote the Zen of Python "Explicit is better than implicit, Simple is better than complex."

gamaanderson, 2015-07-22:
ok, I agree. I just want your vision in this real example:
* I have a netcdf file with fields DBZH and DBZV, pyart read those names directly into radar
* I change the configuration file `reflectivity = 'DBZH'`
* ARTview want to plot those files, he does not know those are reflectivity fields, so calls `get_field_colormap('DBZH')` and `get_field_colormap('DBZV')`, neither one works
* To make it work DEFAULT_FIELD_COLORMAPS need in addiction to key 'reflectivity', the key 'DBZV' and even the key 'DBZH' (even thought I already set `reflectivity = 'DBZH'`)

jjhelmus, 2015-07-22:
In the example, I would expect that the user would need to add a 'DBZV' and 'DBZH' to the DEFAULT_FIELD_COLORMAP dictionary in order to have `get_field_colormap('DBZH')` and `get_field_colormap('DBZV')` return a meaningful colormap.  If these are not supplied I would expect the fields would be plotted using matplotlib's default colormap ('jet') which is what the `get_field_colormap` function will return if the field is not found in the DEFAULT_FIELD_COLORMAP dictionary.

Setting `reflectivity = 'DBZH' ` in the configuration file has no effect on the field names when reading netCDF files which are identified by Py-ART as being Cf/Radial format (most netCDF files) since Py-ART treats these files a bit difference then other files.  There are two reasons for this.  

First, Py-ART does no explicit renaming of field names when reading Cf/Radial files using the current default configuration, see the empty cfradial_field_mapping dictionary in in the default_configuration.py file.    

Second, unlike other read function read_cfradial does not exclude fields which are not mapped to a different name, rather the original field name from the file is used.  If this was not the case no fields would be returned in the radar object.  

An alternative to this is to explicitly provide a list of common field mappings the the cfradial_field_mapping dictionary.  This mapping would need to cover nearly all conceivable field names or else the field would not appear in the radar object which may not be obvious to the user.  In addition if multiple mapped fields are present in the file only the last one would make in into the radar object.  For example if 'DBZH' and 'reflectivity_horizontal' are both mapped to the reflectivity field a file containing both would overwrite one of these fields with data from the other.

If I recall this setup was implemented prior to the introduction of the **file_field_names** argument.  With this argument it may be reasonable to attempt to provide an explicit list of field mappings.  Perhaps with warnings when fields are skipping or overwritten this would be more appropriate.

gamaanderson, 2015-07-22:
Thank you for the explicit explanation, I'm actually pretty happy that netcdf read is not making any mapping and would want to remain that ways, of course I can always force with the `file_fields_name` flag. From the first paragraph if that is the way you think this should work, it is fine for me.

In resume: all my question have been addressed and solved, thank you

jjhelmus, 2015-07-23:
And thank you for asking such good questions!  Thinking about how this configuration should work is catching a number of issues before they even come up.

gamaanderson, 2015-08-04:
with PR #336 merged, can this be closed? or you prefer to let it open?

jjhelmus, 2015-08-04:
Yes this can be closed.  Thanks again @gamaanderson for your work on this.

closed 2015-08-04 by jjhelmus