ARM-DOE/pyart #335: More robust masking procedure for plotting

nguy, 2015-07-21:
I've added a more robust masking option in the form of a dictionary to [my fork](https://github.com/nguy/pyart). It is only added to the plot_ppi currently, as I'd like to get feedback before issuing a PR. 

An example of functionality can be found in this [notebook ](https://github.com/nguy/Notebooks/blob/master/pyart_masking_example.ipynb).

I've been wanting to add this for a while, and now that ARTView is moving ahead I see this as potentially useful for future development there. This way we can just include a plugin or such on that end for pure masking and I believe it will come in handy when editing.

If it does look good, I'll go ahead and implement in the other plot routines and issue a PR.

nguy, 2015-07-21:
One other thing: I'm not a fan of having mask_outside turned on by default. It can confuse data analysis if vmin and vmax are copied from an example or such.
Thoughts?

scollis, 2015-07-21:
I concur.. I was just thinking this today.. I will submit an issue

nguy, 2015-07-21:
I can make this change with the PR if you guys like the masking.

gamaanderson, 2015-07-22:
I like the idea of improving the masking process, but it seems a little confusing, may be is just a question of finding the way of explaining it. An other idea would be to use a [gatefilter](http://arm-doe.github.io/pyart-docs-travis/user_reference/generated/pyart.correct.GateFilter.html#pyart.correct.GateFilter).

scollis, 2015-07-22:
Having pyart.graph.RadarDisplay* accept gatefilter is an excellent idea

nguy, 2015-07-22:
This is a great idea and clearer nomenclature as well!
The code looks equivalent to what I proposed. Probably should have looked harder before coding that up. :)

jjhelmus, 2015-07-22:
You all beat me to the punch, I was going to suggest using a gatefilter.  So +1 from me on that route.

jjhelmus, 2015-07-22:
Also in favor of changing the default behavior of the plotting functions to not mask outside of vmin/vmax.

nguy, 2015-08-27:
I tried something somewhat simple for proof of concept, added gatefilter keyword to ```plot_ppi``` and ```plot_rhi``` and ```_get_data``` functions [here](https://github.com/nguy/pyart/blob/ng-display-gatefilter/pyart/graph/radardisplay.py).

Here's a [quick notebook](https://github.com/nguy/Notebooks/blob/master/pyart_gatefilter_test.ipynb) showing the changes.

Caveat: 
1. Current method permanently changes mask of radar instance. Not sure if this can be avoided.

jjhelmus, 2015-08-27:
@nguy You should use a **GateFilter** instance as an argument to the plot method not the gate_excluded property.  Then _get_data RadarDisplay method is responsibility for extracting out the excluded gates.  

Additionally you cannot count on the field data being a masked array.  Using the following will convert ndarrays to a masked arrays.  This will also preserving any masking on the initial array.

```
if gatefilter is not None:
    mask_filter = gatefilter.gate_excluded[start:end]
    data = np.ma.masked_array(data, mask_filter)
```

We can hack on this at the AMS conference if it can wait until then.

nguy, 2015-08-28:
Good points @jjhelmus . 
I made the suggested changes and pushed. We can continue discussion at conference.
In the meantime, I'm open to any other suggestions, but am thinking this simple change might suffice.

jjhelmus, 2015-08-28:
@nguy Can you make a PR for this branch against ARM-DOE/master? It will make reviewing the changes and eventually merging much easier.  [GitHub Help has a page](https://help.github.com/articles/creating-a-pull-request/) on creating a pull request that provides details on how to do this.

gamaanderson, 2015-08-28:
Nice work @nguy, it would be interesting to add this to all `plot_*` routines, including the ones in `radardisplay_airborne`. It theory it could also be done for grid, but this would include developing a `gatefilter_grid` or something like that

nguy, 2015-08-28:
I wanted to get it out there to see if the changes are agreeable before modifying the other plot routines. But I would update those before merger. 

I agree with this being developed to grid further down the road.

nguy, 2015-08-28:
PR #363 submitted. By the way, what does it mean when the checks fail?

jjhelmus, 2015-08-28:
It means when the continuous integration service build Py-ART from source after merging the proposed changes, some of the unit tests failed.  You can expand the check and click on the Details link to see what is causing the failing tests.  In this case the RadarMapDisplay.plot_ppi_map method is making a call to RadarDisplay._get_data using the older call signature without the gatefilter which causes the test to fail.

nguy, 2015-08-28:
Ah, I didn't look to see about that. Do you want me to update that? 
I could also add into RadarDisplay_Airborne (and maybe change the name).

nguy, 2015-08-28:
I pushed those updates and now the checks have passed.

Slightly off topic question: What ramifications would changing ```pyart.graph.radardisplay_airborne.RadarDisplay_Airborne``` to ```pyart.graph.radardisplay_airborne.AirborneRadarDisplay``` have?

jjhelmus, 2015-08-28:
Changing **RadarDisplay_Airborne** to **AirborneRadarDisplay** should probably go through a depreciation cycle (so the old name still works for at least one point release) but that can be accomplished by creating an alias in the `__init__.py` file.  Other than that some imports and documentation would need to be change but I'm in favor of the name change.

nguy, 2015-08-28:
Should this be done with the submitted PR or in a separate PR?

nguy, 2015-08-28:
I may try to add a reader for Wyoming Cloud Radar in the future. I could hold off the change until that time if you like.

gamaanderson, 2015-08-28:
I should point it out that it would need some thinking and it is not a easy decision, but it would be possible to make `RadarDisplay` accept airborne radars.

dara74k, 2015-08-28:
hi Nick,
it took me 1 whole day to successfully install pyart(Anaconda pkg) in linux
(and overcoming a whole pile of issues).
when i copy a code from the pyart-examples website and paste it in the line
of the ipython notebook, i get errors. Can you be kind enough to guide me
through this problem?

thank you in advance
best regards,
Diar

nguy, 2015-08-28:
I'm trying to remember the exact issues, but when I added the separate class, it was because there were some fundamental differences with the data structure and files at the time. This might be able to be revisited now that cfradial and Sigmet files are used by the aircraft  and @jjhelmus has done an awesome job to update the read routines to support extended headers, etc. 

And the elephant that I rarely acknowledge is that the class *only* supports tail Doppler radar data. Other airborne radar types cannot be read and/or visualized with this class currently. That is why the name change may be adopted. That way, there could be a call back to other components for different radar types (for example the belly radar on some aircraft behaves more like a surface-based system, while other aircraft have up to 6 antenna!)

jjhelmus, 2015-08-28:
Probably should go in a separate PR, typically you want to keep a PR to fixing a single issue.  Sometimes it is a pain but it makes rolling back changes easier when you make merges atomic.

nguy, 2015-08-28:
You got it @jjhelmus ! It should be ready to go now after the checks complete.

jjhelmus, 2015-08-28:
@gamaanderson I think it could be possible to combine **RadarDisplay** and whatever we call the airborne display class.  The whole graph sub-module could use a little more care and attention although it is a lot better thanks to your recent changes.

nguy, 2015-08-28:
@dara74k Post the issues you are having in the [Users group](https://groups.google.com/forum/#!forum/pyart-users). Also make sure to include the errors to help the community troubleshoot.

nguy, 2016-01-07:
Can this be closed?

jjhelmus, 2016-01-08:
Yes, this was address in #363, closing.

jjhelmus, 2016-01-08:
Thanks @nguy for the follow up, always good to close issue that have been fixed.

closed 2016-01-08 by jjhelmus