ARM-DOE/pyart #345: Despeckling of radar and grid data

jjhelmus, 2015-08-04:
The ability to remove salt/pepper noise (despeckling) from radar and grid fields should be added to Py-ART.  Adding this function to the GateFilter would also be reasonable.  @kirknorth implemented a despeckle grid filter in #133 which may serve as a base for this work.

gamaanderson, 2015-08-05:
+1 for this suggestion, the interpolation methods of Barnes and Kressman are extremely sensible to speckle. So despeckling before mapping, would be recommended.

tjlang, 2015-08-05:
If folks need despeckling RIGHT NOW, it is implemented in the csu_misc sub-module of CSU_RadarTools (https://github.com/CSU-Radarmet/CSU_RadarTools). It's a simple copy of what was done in solo. It uses loops so it is very slow and I am sure whatever gets done in Py-ART will be better. It also requires you to pull your data out of the Py-ART radar object and put it into separate ndarrays. But it is working right now and there is even a section on how to use it near the bottom of the CSU_RadarTools demo notebook (https://github.com/CSU-Radarmet/CSU_RadarTools/blob/master/notebooks/CSU_RadarTools_Demo.ipynb).

mvanlierwalq, 2015-08-05:
Another option could be a grey opening/closing operation. Opening is erosion followed by dilation, and closing is the opposite. Scikit-image has these functions http://scikit-image.org/docs/dev/api/skimage.morphology.html and I've used them with some success. One possible issue is that I think these methods are designed for cartesian coordinate systems, not polar. Also, I don’t know the relative merits compared with what is typically used. Just a suggestion.

kirknorth, 2015-08-06:
I have used similar techniques on PPI and VPT radar volumes that @mvanlierwalq points out. Note that binary dilation and erosion methods (or more generally mathematical "morphology") are available directly from SciPy under the `ndimage` module. I have generally had success with these methods as well, however I have never fully addressed the periodic boundary introduced by 360 deg PPI volume scans. @jjhelmus you may have some pointers for this since your new dealiasing routines are able to properly handle these types of scans.

jjhelmus, 2015-08-06:
The periodic boundary of PPI and other boundary effects are going to be tricky.  The morphology and label functions in SciPy's `ndimage` sub-package do not deal with these issues.  The filter function in `ndimage` do have a **mode** parameter which specified how boundaries are handled which might be helpful to see how those are implemented.  In the region dealiasing routine I addressed the periodic boundary by allowing edges between regions on either side of the boundary.  This is not perfect, the preferred method would be to allow regions to span the boundary, but it seems to work well enough.

I'm thinking a quick and dirty despeckling routine which uses **ndimage.label** and removes 'small' regions would be a good start.  From there some of these more difficult issues can be resolved as needed.  

Can anyone think of a use case where this would not be used on a GateFilter instance?

tjlang, 2015-08-06:
FYI, traditionally in old software like solo, etc., despeckling was based on the number contiguous gates in a ray. I always felt that adjacent rays should be considered too (e.g., the 2D approach folks are suggesting here with ndimage, etc.), but one unexpected side benefit of the 1D approach is that it nukes sidelobe much better than expected. For example, in DYNAMO the TOGA radar had a fairly large beamwidth which led to significant sidelobe. Sometimes when we had skin paints on other ships you even saw a thin circle surround the radar. A 2D image analysis would save that from despeckling, but 1D does not. Just wanted to throw that out for people to consider.

gamaanderson, 2015-08-06:
To address the wrap_around couldn't we just copy to some initial rays to the end, and remove them afterward ? 

What do you mean by using GateFilter? the despeckling wouldn't change the data, just create a filter? That sounds ok, just remembering that some times the user may need to apply the filter to permanently change the data, for instance GateFilter is not implemented to mapping, graphing and saving routines.

As @tjlang point it out, we may want more than just one despeckling method; closing, 1d, 2d could all coexist. It is just a question of which ones we spend time to implement.

stechma2, 2015-09-19:
Some time back I wrote some fortran routines to handle despeckling and defreckling, which I then converted into python modules using f2py. I did not reference the manner in which solo handles these procedures, but I expect they may be similar. I did however perform direct comparisons between solo and my code, and the results are nearly identical (see the comparison below, which also includes a few additional edits such as thresholding on NCP).

I implemented a freckle average threshold parameter which allows for the definition of a freckle to be easily changed (by considering the average of adjacent gates as compared to the target gate). I had plans to also implement the ability to easily redefine the maximum speckle size, but never got around to it (currently hard coded to a max of 3 consecutive gates).

Currently, both 1d and 2d approach are used for both the despeckling and defreckling routines. The 1d just looks at adjacent gates in each ray. The 2d considers the same gate number as the target, but in adjacent rays (so not truly 2d, but on the right track).

I have the intention of making the code far more efficient (it currently relies heavily on loops), and to improve the 2d algorithms, but am currently too busy with other projects.

One of the parent python scripts and all of the fortran code (and resultant modules) are available here: https://uofi.box.com/s/2oksf6npr5eac5mq9x22lnrc8ymtc0l0

The script is reasonably commented and/or self-explanatory, but please let me know if anything is unclear if you choose to adopt any of this.

[image]

tjlang, 2015-09-21:
I recently sped up the 1D despeckling in csu_radartools.csu_misc, via scipy.ndimage. It works on either a single ray, or loops thru a set of rays. It's pretty fast now. There is no interrogation of adjacent rays, but as I mentioned previously, this approach has its merits. There's not much code involved so it would be easy for someone to port into Py-ART.

https://github.com/CSU-Radarmet/CSU_RadarTools/blob/master/csu_radartools/csu_misc.py

jjhelmus, 2015-09-21:
Thanks for the pointers to the code @tjlang  and @stechma2.  I'll have a look and hopefully find time to implement some despecking/defreckling routines in Py-ART in the near future.  I'll be in contact if I have any questions.

tjlang, 2016-05-19:
I've got a bead on some code that will address the 2D despeckling issue. It uses ndimage.label and accounts for the periodic boundary issue in 360 PPIs, which is easier to solve than you might think. Basically, append a sweep to itself, label objects, relabel the objects along the right edge of the first sweep copy with adjacent values on the left edge, then remove the duplicate sweep. Here is an example that shows deletion of additional pixels that were missed during the 1D despeckling approach in CSU_RadarTools. Note that user can specify maximum size to consider a speckle. 

[image]

The great part is that you can also use the engine to find cells in sweeps, and port the labels to the radar object as a separate field. You can even apply a smoother first to avoid over-diagnosing objects. Here are objects greater than 40 dBZ in the scan from above:

[image]

I need to clean up and document the code, then figure out the best way to port it into Py-ART. But I wanted to let folks know this is coming down the pipe

jjhelmus, 2016-05-19:
Those results look really good @tjlang, I'm excited to see this get added to Py-ART.  Are you planning on supporting RHI scans also?

scollis, 2016-05-19:
+1 to @jjhelmus very funny we were just discussing this with respect to mesh gridding in our research group yesterday!

Dr Scott Collis
Purveyor of lofty observations (Atmospheric Scientist) 
Environmental Science Division
Argonne National Laboratory
Mb: +1 630 235 8025
Of: +1 630 252 0550
Become a Py-ART user today! http://arm-doe.github.io/pyart/ <http://arm-doe.github.io/pyart/>

gamaanderson, 2016-05-19:
@CesarBeneti you may be interested in this. This despeckling looks really good. We will also be working in adding this to [artview](https://github.com/nguy/artview).

CesarBeneti, 2016-05-19:
Hi Anderson,
I will have a look on that. Thanks for info.
Regards

Cesar Beneti
Diretor Executivo
SIMEPAR - Sistema Meteorológico do Paraná
Curitiba - PR
Tel: (41) 3320-2002

"Antes de imprimir pense em sua responsabilidade e compromisso com o MEIO
AMBIENTE"

tjlang, 2016-05-19:
Yes, @jjhelmus, RHIs should be supported. The 360 PPI adjustments are only called under certain circumstances. Without those, the 2D labeling is very similar to what you might do naively without any consideration for periodic boundaries, plus it operates on a sweep-by-sweep basis.

nguy, 2016-05-20:
This is great! @tjlang this looks similar to the RadStats program from back in the day at CSU. A step toward TITAN capabilities @scollis ?

The despeckling would be a great addition for ARTview as well, I know it's been inquired about more than once if we plan to give a SOLO alternative.

scollis, 2016-05-24:
So, what can we do to help? If you need help documenting etc let us know.. I need some code like this :)

tjlang, 2016-05-24:
Give me to the end of this week to put together a PR. I need to look into how best to merge this within the `GateFilter` approach that Py-ART uses.

scollis, 2016-05-24:
Perhaps have it return a gatefilter object.. Or if it just creates a field with binary values it is very easy to create a gatefilter from that

nguy, 2016-05-24:
I like the idea of *despeckle.py* housing the different despeckling routines and returning a GateFilter object. The trick may be merging the GateFilter object with existing? But this might be done already internally, I don't recall.

tjlang, 2016-05-25:
Despeckling addressed by this PR: https://github.com/ARM-DOE/pyart/pull/538

closed 2020-03-26 by zssherman