ARM-DOE/pyart #347: Document special dictionary keys

jjhelmus, 2015-08-05:
PR #266 introduced a number of "special" dictionaries keys which change how data is written out to disk in netCDF files.  These keys should be documented in the **Radar** and **Grid** classes.

jjhelmus, 2015-10-01:
I think until these special keys are used by other function a more appropriate place to document them is in the `write_grid` and `write_cfradial` functions.  Is that acceptable @gamaanderson?

gamaanderson, 2015-10-01:
I think it is acceptable, but I would also see the Radar and Grid classes as possible places for documentation. For instance, see what I have done with the [Points](https://github.com/nguy/artview/blob/master/artview/core/points.py#L9-L51) class in artview

jjhelmus, 2015-10-01:
I see a case for adding information about these keys to either the Radar and Grid classes or the write functions but I do not like the idea of having the information replicated in four places (two is enough already).  

I tend to think users would think to look at the documentation of the write functions for details on how to change the write behavior not the documentation of the container holding the data, but I could be wrong.  Therefore I added to the docs for the write_grid and write_cfradial functions in ac497f5.  

If we find that users are missing this information and it should be added in another location I think we can revisit this issue.

closed 2015-10-01 by jjhelmus

gamaanderson, 2015-10-01:
I understand your point.

My problem is that I see things different, a function can't make "demands" or "add specifications" over a class, it can only respond to the class documentation. I also, I don't believe in the existence of "abstractions"; data is data, and a class is data+"functions to operate the data". Writing is either saving the data to the disc, or converting from one data convention to an other data convention.

Anyway, this is just my opinion, you have an other one and you must write the code following your views, not from anyone else.

commit ac497f50 (2015-10-01): DOC: Document special keys which effect writing of netCDF files

closes #347