ARM-DOE/pyart #355: Using dealias_region_based with NEXRAD level 2

seberry2, 2015-08-11:
I am trying to use dealias_region_based on some NEXRAD level 2 files. However, the following error is returned:

Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "C:\Anaconda\lib\site-packages\pyart\correct\region_dealias.py", line 407, in _combine_regions
    status, extra = edge_tracker.pop_edge()
  File "C:\Anaconda\lib\site-packages\pyart\correct\region_dealias.py", line 643, in pop_edge
    edge_num = np.argmax(self.weight)
  File "C:\Anaconda\lib\site-packages\numpy\core\fromnumeric.py", line 943, in argmax
    return argmax(axis)
ValueError: attempt to get argmax of an empty sequence

I am also working with some cfradial files and the dealiasing works fine. 
This leads me to believe (I'm not to certain on this conjecture) there is some discrepancy in the way Py-ART reads in the two files which causes the error.

Can anyone else replicate this? Is there a simple fix?

Here is a link to the two files I am working with:
https://drive.google.com/file/d/0B9eYWp9eQxNxcVU5VE9oNWxfREU/view?usp=sharing
https://drive.google.com/file/d/0B9eYWp9eQxNxLXM3MThXR1ZmQVE/view?usp=sharing

Thanks!

jjhelmus, 2015-08-11:
@seberry2 What version of Py-ART are you running?  There were some bugs in the region based dealiasing routine which would cause errors like the one you describe when two or fewer regions were found in a sweep.  These have been fixed, PR #327, but may not have made it into the version of Py-ART you are running as they have not made it into a release.

Running to following command from the command line should provide the Py-ART version that is available, `python -c "import pyart; pyart._debug_info()"`

seberry2, 2015-08-11:
I am running Py-ART version 1.3.0.

scollis, 2015-08-11:
Yeah.. @jjhelmus and I encountered this with SAILS nexrad files.. some 
tilts have all _FillValue for radial velocity
A quick workaround is here
http://nbviewer.ipython.org/github/scollis/notebooks/blob/master/Radar%20demo%20IL%20Torn.ipynb
First few cells only keeping good sweeps

jjhelmus, 2015-08-11:
The bug you are running into has been fixed in Py-ART but you will need to update to the latest version to obtain this bugfix.  Using the latest version from GitHub I was able to use the region based dealiasing routine on the KDIX20131218_090423_V06 file without issue:

[image]

jjhelmus, 2015-08-11:
You should be able to upgrade to Py-ART version 1.4.0 using conda as there are binary files available for that version on my [binstar.org channel](https://binstar.org/jjhelmus/pyart) but I do not think these bug fixes you require are in this release.  We should be making another release later this month or early next month in preparation for the AMS Radar meeting.

seberry2, 2015-08-11:
Using the suggestion from @scollis, I got it to work. I'll update the version of Py-ART I am running when the new release is available. Thanks for your help!

jjhelmus, 2015-08-11:
Glad we could be of help.

closed 2015-08-11 by jjhelmus