ARM-DOE/pyart #356: Data model for radar spectra

scollis, 2015-08-11:
Discussing with @mvanlierwalq about spectra calculations makes me think we need a thread to talk about a data model in Py-ART for Spectra

Could it be as simple as letting the data array be 3D? A spectra object has all the normal metadata but two arrays (vert, horizontal) or one array  of Power(velocity, time, range)?
and also has a spectra.velocity dictionary defining the axis.

scollis, 2015-08-11:
I will also talk with Mike Dixon to see if he has considered a file format, CF extension for CF-Radial

jfigui, 2016-08-25:
Be aware that in some cases the number of pulses used to compute the spectra may vary slighly from ray to ray (such as if you just collect all the pulses received within a given azimuth span).
Also, if you have multiple PRF the Doppler velocity span will vary.

hhuangmeso, 2016-08-25:
@scollis Actually, cfradial v2.0 draft considers the IQ/spectrum data. See this [cfradial 2.0](https://github.com/NCAR/CfRadial/blob/master/docs/CfRadialDoc.v2.0.draft.pdf)

scollis, 2016-08-25:
Yep @hhuangmeso in fact the spectra definition may be backported to 1.4. 2.0 uses netCDF groups which we need to think carefully before supporting. 

I'll be working with some doppler spectrum as part of some ARM work I am doing in the next few months so this will allow me to think a little more on a data model. It will be heavily based on CF-Radial.. But we also want it to interact well with Python (ie insofar as possible allow easy vectorization  )

josephhardinee, 2016-08-25:
We have already placed a spectra definition into cf radial 1.4 . As Scott mentioned, we will probably hold off on moving to cf radial 2 for at least a few years.

hhuangmeso, 2016-08-26:
Since, the IQ/spectrum data are large, which make it necessary to consider the compression/scaling carefully. I've already implemented the basic part to convert the IQ to the radar variables. While, it's mostly an engineering issue. Some details like clutter detection/elimination are not easy for me. If necessary, I can contribute to this work.

mgrover1, 2022-11-07:
Merging with #670

closed 2022-11-07 by mgrover1