ARM-DOE/pyart #364: Change Airborne RadarDisplay class

nguy, 2015-08-28:
The following change is suggested: ```pyart.graph.radardisplay_airborne.RadarDisplay_Airborne``` to ```pyart.graph.radardisplay_airborne.AirborneRadarDisplay```. In a side discussion in #335, @jjhelmus noted:
"Changing RadarDisplay_Airborne to AirborneRadarDisplay should probably go through a depreciation cycle (so the old name still works for at least one point release) but that can be accomplished by creating an alias in the __init__.py file. Other than that some imports and documentation would need to be change but I'm in favor of the name change."

As this class was originally created for the NOAA P-3 Tail Doppler Radar (TDR) it is somewhat specific to this type of radar system. Another possibility to is to rename the existing class **AirborneRadarDisplayTDR** and also include a **AirborneRadarDisplayLF** that would call a regular **RadarDisplay** for the lower fuselage radar, which behaves more like a surface-based system in terms of scanning and display properties.

jjhelmus, 2015-09-01:
I vote for renaming the Airborne class to **AirborneRadarDisplay**.  I would like to release version 1.5.0 prior to this change, but I can make this change after the release if you would like.

nguy, 2015-09-01:
That sounds good. I have to take care of some other things before thinking about further airborne radar additions. Question to chew on: What do you think about a plotting class for time series radar data such as upward-pointing lidar or cloud radar, etc?

josephhardinee, 2015-09-01:
A few quick things, those plots would be time height displays. Time series implies something else. Before we talk about the plotting, it is worth noting pyart doesn't actually have the capability to support reading these types of files yet. I believe Scott initiated a discussion about this in another thread, as it requires adding a new dimension(Doppler velocity) to the data model plus possibly addressing a few other issues.

scollis, 2015-09-01:
What we are dealing with here is a clash of nomenclatures.. In the rest 
of the world a time series is anything with datetime x-axis.. so 
technically a time height plot is a class of time series plots.. But 
yes.. "Time series data" has a special meaning in the radar world.

kirknorth, 2015-09-01:
@josephhardinee there is currently one module to read ARM vertically pointing cloud radar data, specifically the KAZR *a1* datastream, see [`arm_vpt`](https://github.com/ARM-DOE/pyart/blob/master/pyart/aux_io/arm_vpt.py) and Issue #281.

josephhardinee, 2015-09-01:
Oh nice, I wasn't aware that had been added.

jjhelmus, 2015-09-01:
Py-ART can plot "time/height" radar data.  This is interpreted as a VPT scan and the **RadarDisplay** class has a `plot_vpt` method.  This could be extended to lidar or other similar radars.  Is this sufficient or is a different format plot required? 

As for plotting radar spectra or true radar time series data (I and Q data), that is another discussion which should be discussed in #356.
 
As for reading, Py-ART can read the "a1 level" ARM SACR cloud radar files, including those from VPT scans as these are in CfRadial format.  Files from ARM's KAZR radar can be read with the `pyart.aux_io.read_kazr` funtion.  Support for reading other zenith points ARM radars is an open issue, #270

jjhelmus, 2015-09-01:
Added note in #366 to add an example of plotting VPT data to the documentation.

kirknorth, 2015-09-01:
As @jjhelmus pointed out, there currently is a `plot_vpt` method in `RadarDisplay`, however from my experience it does need to be revisited. For example, the time axis output by plot_vpt is simply an index, e.g., line [563](https://github.com/ARM-DOE/pyart/blob/master/pyart/graph/radardisplay.py#L563) or line [565](https://github.com/ARM-DOE/pyart/blob/master/pyart/graph/radardisplay.py#L565). Furthermore, there is no time limits functionality, so plotting a KAZR a1 file can take quite a bit of time since these files are 24 hour records with over 23,000 profiles.

scollis, 2015-09-01:
Hmm... A tool in Py-ART to add sweeps to a VPT would be very handy (eg: 
radar.add_sweeps('hourly')) this could allow users to then break up VPTs 
into discrete blocks which would help for analysis as well as plotting

jjhelmus, 2015-09-01:
VPT plotting ignores all sweep data, so adding sweeps to this data will not accomplish the desired goal.  Adding time awareness to the  `plot_vpt` method and adding parameters to specify limits may be better.

nguy, 2015-09-01:
Sorry for the confusion with time series, yes I was thinking of a time height plot (but it is a time series as Scott points out). I think it's important not to lose sight that these types of plots are used in conjunction with the more classic single dimension time series plots. But I digress...

I had not thought of adapting the **vpt** routines, but agree with @kirknorth that these may need to be revisited for time indexes. I did something like this [here](https://github.com/nguy/AWOT/blob/master/awot/graph/radar_vertical.py#L916), which calls [this routine](https://github.com/nguy/AWOT/blob/master/awot/graph/common.py#L255). It may be a bit obtuse but I think it's probably similar to what Pandas does behind the scenes for formatting (I haven't checked this)?

+1 @scollis  for ```add_sweeps_to_vpt```

nguy, 2016-02-04:
Can this be closed, or are there some conversation pieces that need to be kept alive?

jjhelmus, 2016-02-04:
Yes, closing.  If improved VPT plotting is desired a new issue should be opened on that topic.

closed 2016-02-04 by jjhelmus