ARM-DOE/pyart #366: Additional examples in documentation

jjhelmus, 2015-09-01:
Additional examples should be added to Py-ART's documentation to describe new features which have been added to the package.

This list will be updated as suggestions are made:

* Dealias with region based algorithm.
* Dealias with unwrap_phase algorithm.
* Plot VPT data
* GateFilter usage.
* Create a Grid with the gates_to_grid algorithm

Rumpkie, 2016-02-26:
I'm going to take a stab at this so can you assign #366 to me.
-C

Rumpkie, 2016-03-04:
@jjhelmus, Did you have notebooks with some of these already created that I could draw from?

jjhelmus, 2016-03-04:
I think a few examples can be found in the notebooks from the [2015 AMS](https://github.com/openradar/AMS-Short-Course-on-Open-Source-Radar-Software) and [2015 ARM/ASR meeting](https://github.com/ARM-DOE/notebooks/tree/master/ASR_PI_2015) course materials.  I know that both have examples of the region based dealiasing algorithm.

zssherman, 2020-03-26:
Examples for most of these exist. I also added more notebooks for ingest and gridding etc. Closing as this is outdated. Can open a new issue if new example ideas arise.

closed 2020-03-26 by zssherman