ARM-DOE/pyart #367: Sigmet: Units conversion for 1-byte spectrum width

deeplycloudy, 2015-09-01:
According to the comments in the source, which are consistent with the IRIS docs, the 1-byte spectrum width values should be multiplied by the Nyquist velocity.

 https://github.com/ARM-DOE/pyart/blob/master/pyart/io/_sigmetfile.pyx#L709

This doesn't happen automatically, although according to line 705, a similar multiplication is done to velocity in get_data (actually, in read_data, line 216).

Would the maintainers be amenable to replicating the VEL method for WIDTH so that the conversion is done for the user?

deeplycloudy, 2015-09-01:
Thanks, @jjhelmus! Let me know if I can help.

jjhelmus, 2015-09-01:
I believe PR #369 should address this, can you verify @deeplycloudy ?

deeplycloudy, 2015-09-01:
The correction factor used in PR #369 is unambiguous range, but should be unambiguous velocity - i.e., I think the nyquist used in the VEL correction.

jjhelmus, 2015-09-02:
Sorry about that, I saw unambiguous in the IRIS Programmers Manual and jumped to unambiguous range but you are correct the unambiguous velocity (twice the Nyquist) should be used.  I updated PR #369 to reflect this.

deeplycloudy, 2015-09-04:
I did a comparison among the following:
1. read_rsl, with a .raw
2. read_rsl, with a .uf created by the IRIS system
3. read_sigmet 
The resulting plots are attached. If you ignore the bad rays in the read_sigmet version, you can see that the spectrum width calculated by this pull request is double that reported by RSL. 

I agree that this pull request correctly implements the span of unabiguous velocities (twice the Nyquist), but RSL seems to be using the Nyquist. Since the UF was converted by IRIS, and both RSL versions are consistent, I'd tend to favor using the Nyqvist, but the meaning is admittedly unclear in the IRIS manual.

[image]
[image]
[image]

jjhelmus, 2015-09-08:
@deeplycloudy I checked into the method that RSL and RadX are using to convert 1-byte WIDTH data and both programs scale by the Nyquist.  From the text in the IRIS Programmer's Manual this does not seem correct but given that both of these use this conversion it should be used in Py-ART.  I created PR #372 which scales the data by the Nyquist.  I will merge shortly and close the other PR.  Thanks for reviewing this issue.

closed 2015-09-08 by jjhelmus

jjhelmus, 2015-09-22:
FYI had a ARM engineer run a Sigmet/IRIS file with a 1 byte WIDTH field through the **productx** program available from the vendor.  It agrees with the current implementation in Py-ART of the unambiguous velocity being equal to the nyquist velocity.

njmessina, 2019-03-27:
Hi there...
so I'm having this same problem with pyart... my velocities from an IRIS file vary from -1 to 1.  I'm sorry I'm new at this, but does anyone have any advice as to how I should handle this?
thank you!!
Joe

PR #372 (2015-09-08): BUG: 1-byte width field from Sigmet files returns in m/s
closes #367
commit 1d492c0d (2015-09-08): BUG: 1-byte width field from Sigmet files returns in m/s

closes #367