ARM-DOE/pyart #384: Bug in aux_io.radx.read_radx

tjlang, 2015-09-24:
I have been attempting to use aux_io.radx.read_radx to read current Level 2 NEXRAD files, in order to save a step in converting them to CF Radial myself first. (This is mostly to simplify the split-cut scan issue.) Anyway, the reader consistently fails, I think because Radx is putting the file the wrong place and not naming it right. See below for an example.

In [2]: pyart.aux_io.radx.read_radx('/raid1/tlang/l2/KGLD/KGLD20140605_061948_V06')
WARNING - NexradRadxFile::_handleAdaptationData
  Buffer too small, size: 2404
  Should be at least: 8808
  Ignoring adaptation data
WARNING - NexradRadxFile::readFromPath
  Adaptation data probably not set, ignoring
  File: /raid1/tlang/l2/KGLD/KGLD20140605_061948_V06

---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-2-d30d1e4b597c> in <module>()
----> 1 pyart.aux_io.radx.read_radx('/raid1/tlang/l2/KGLD/KGLD20140605_061948_V06')

/raid1/tlang/anaconda/envs/py27/lib/python2.7/site-packages/pyart/aux_io/radx.pyc in read_radx(filename, **kwargs)
     45              '-outdir', head, '-outname', tail, '-f', filename])
     46         print('debug', head, tail, tmpfile)
---> 47         radar = read_cfradial(tmpfile)
     48     finally:
     49         pass

/raid1/tlang/anaconda/envs/py27/lib/python2.7/site-packages/pyart/io/cfradial.pyc in read_cfradial(filename, field_names, additional_metadata, file_field_names, exclude_fields, delay_field_loading, **kwargs)
    120 
    121     # read the data
--> 122     ncobj = netCDF4.Dataset(filename)
    123     ncvars = ncobj.variables
    124 

/raid1/tlang/anaconda/envs/py27/lib/python2.7/site-packages/netCDF4.so in netCDF4.Dataset.__init__ (netCDF4.c:21986)()

RuntimeError: NetCDF: Unknown file format

What happens is where I am running Py-ART, the temp file is created (in this case named tmprcqgph.nc). It is 0 bytes. Then Radx will create a subdirectory of the current Py-ART directory using the original L2 file's date, and then it creates a valid CF Radial file (in this case named cfrad.20140605_062048.057_to_20140605_062509.317_KGLD_v750_SUR.nc) in that subdirectory. Py-ART doesn't appear to know where to look for the valid file. It wants to read /usr/people/tlang/pmm/iphex/20140605/tmprcqgph.nc (in this example), which exists but is 0 bytes, so it just fails with a read error. 

This is on a Linux system but I have noticed similar issues in Mac OS X.

jjhelmus, 2015-09-24:
What version of Radx are you using?  The `-outname` flag was added in a recent release.  One way to check is to see if the output of `RadxConvert -h` includes lines mentioning that switch.

tjlang, 2015-09-24:
I just upgraded to the latest Radx and that fixed it! Sorry for the trouble.

closed 2015-09-24 by tjlang

scollis, 2015-09-24:
All good.. perhaps we should add a warning or test to make sure the 
version is correct

jjhelmus, 2015-09-24:
As far as I know there is not a method for querying Radx on its version.

tjlang, 2015-09-24:
Yeah, I noticed that in the Radx documentation & help ...

tjlang, 2015-09-24:
If a warning or print statement is added, I note no issues with the 20150826 Radx build. Prior to that, as recent as an early July build, it wasn't working properly with read_radx().

jjhelmus, 2015-09-24:
PR #385 will raise an IOError with a note suggesting upgrading to the latest version of Radx when no file is created when using RadxConvert.