ARM-DOE/pyart #386: Drop support for Python 2.6

jjhelmus, 2015-09-24:
Py-ART contains a number of kludges that are requires to support Python 2.6. Examples include [a lack of dictionary completions](https://github.com/ARM-DOE/pyart/blob/9088d22249f1952dd3665c703bb2e2710bfcc7f2/pyart/util/xsect.py#L108) and [no total_seconds datetime attribute](https://github.com/ARM-DOE/pyart/blob/9088d22249f1952dd3665c703bb2e2710bfcc7f2/pyart/io/mdv_common.py#L1066).

Dropping support for 2.6 would allow for simplier code and less work for future development.  Does any of out users have a continues need for Python 2.6 support?  I know @kirknorth used have a need for Python 2.6 support but I am not sure of his current requirements.  Please speak up now or format support will be dropped.

gamaanderson, 2015-09-24:
Last time I checked @pfhein was also using python 2.6.

kirknorth, 2015-09-24:
@jjhelmus I have had a virtual environment running 2.7 for quite some time now.

pfhein, 2015-09-24:
@jjhelmus, @gamaanderson, I have recently moved to 2.7 though I still have 2.6 available.  I am okay with ending Python 2.6 support.  Anaconda made it easy to upgrade my clone RHEL 6 python 2.6, and there are "official" RHEL software collections of Python 2.7 too.  So I do not see the need to support 2.6.

jjhelmus, 2015-09-25:
How does the following sound:

The next Py-ART release, version 1.6.0, will be the last release to support Python 2.6.  Once the release is made and the 1.7.0-dev cycle is started, I'll remove 2.6 from the continuous integration matrix and replace some of the 2.6 specific code with more idiomatic code.  

This gives anyone running 2.6 a few more months to upgrade.  Once release they can pin version 1.6.0, if there is a need bug fixed in the mainline can be merged into 1.6.x releases that will continue to support Python 2.6.

scollis, 2015-09-25:
Sounds reasonable to me

pfhein, 2015-09-25:
I like the plan.  Python 2.6 is hard to support and upgrades to 2.7 are available.

PR #479 (2016-02-05): MAINT: Remove support for Python 2.6 and 3.3, add Python 3.5
closes #386
closed 2016-02-05 by jjhelmus