ARM-DOE/pyart #389: Phase processing in python3

swnesbitt, 2015-10-01:
I have been slowly upgrading to python 3 and ran into a snag central doing phase processing with PyART.  AFAIK, cylp is not compatible with python 3, and neither is PyGLPK (last updated last decade).  I am currently dealing with the pain of installing CVXOPT on OS X (BLAS/LAPACK) using homebrew.  It would be really nice if cylp could be upgraded to python 3 (and support more recent versions of cbc).

jjhelmus, 2015-10-01:
Agree completely @swnesbitt .  The lack of a Python 3 compatible LP library is what is holding me back from changing my default Python environment to Python 3.  I do not want to change until all of Py-ART unit tests run in Python 3.4 and none are skipped.  The CyLP folks do have an open issue coin-or/CyLP#15 on Python 3.4 support.  

The [ecyglpk package](https://github.com/equaeghe/ecyglpki) might be worth a look for a more modern Python interface to GLPK.  Not sure if it supports Python 3.

scollis, 2015-10-01:
If anyone wants to put some work into this it would have a great impact 
on Py-ART! We need a better option for LP

jjhelmus, 2015-10-01:
I was able port enough of CyLP to Python 3 to allow the unit tests in CyLP and Py-ART to pass.  I'm not going to provide link to the branches since they are a mess and need to be cleaned up before submitting and PR to CyLP. 

What is the issue with CyLP not support recent version of Cbc, @swnesbitt?  I am still using 2.8.8 and have not tried the 2.9.x release.

swnesbitt, 2015-10-01:
I thought I had a build issue with 2.8.5 on OS X 10.11, but will try 2.8.8

jjhelmus, 2015-10-02:
Pull request submitted to CyLP to add support for Python 3 coin-or/CyLP#28.  Minor modification are needed for Py-ART's phase processing to work with CyLP in Python 3.  Changes submitted as #392.  

I've been testing with CBC 2.8.8 without issue.  I have a conda package for the library build for [OS X and Linux 64](https://anaconda.org/jjhelmus/cbc).

jjhelmus, 2016-03-16:
I found a [fork of pyglpk](https://github.com/bradfordboyle/pyglpk) which supports newer versions of the GLPK library and in the v0.4-dev branch supports Python 3!  Using this and CyLP with my PR applied I was able to get all the Py-ART unit tests to pass in Python 3.5 on my Linux machine.  I believe the same should be possible on OS X.  

The tests emit a few warnings including #478 which I want to clean up.  I'll work on creating conda packages of the various phase processing modules to make this more robust and reproducible.

scollis, 2016-03-16:
Fantastic!

hhuangmeso, 2016-04-12:
@jjhelmus Would you please make a pyglpk/cylp package which could be easily installed on OSX and linux for python 3 on anaconda cloud? That'll help a lot for us to move to python3. Thank you!

hhuangmeso, 2016-04-12:
I found cvxopt/cvxopt_glpk packages https://anaconda.org/omnia/cvxopt/https://anaconda.org/mlgill/cvxopt_glpk which seem working well in my OSX. The accuracy of the results is still under test. By the way, linprog has been added to scipy since 0.15, which would benefit the portability of pyart.

scollis, 2016-04-12:
Very interesting! Will definitely need to check out linprog

Dr Scott Collis
Purveyor of lofty observations (Atmospheric Scientist) 
Environmental Science Division
Argonne National Laboratory
Mb: +1 630 235 8025
Of: +1 630 252 0550
Become a Py-ART user today! http://arm-doe.github.io/pyart/ <http://arm-doe.github.io/pyart/>

jjhelmus, 2016-04-13:
> Would you please make a pyglpk/cylp package which could be easily installed on OSX and linux for python 3 on anaconda cloud?

I'm been working on this but creating portable packages for pyglpk, cylp and cvxopt with glpk support is turning out to be more of a channel that I had expected.  Part of this is that there is not release of cylp and PyGLPK which supports Python 3 so I need to build against a fork.  The other difficult is that cvxopt and cylp both require linking to BLAS/LAPACK which is a bit difficult to do in conda in a portable manner.  I'll keep working on these but it may take a bit.

scollis, 2017-07-13:
So @jjhelmus 's PR is still languishing over at CyLP
https://github.com/coin-or/CyLP/pull/28

I stumbled on this:
https://github.com/coin-or/pulp
I wonder if it will help us.

zssherman, 2020-03-26:
The python 3 branch was merged into CyLP master, closing.

closed 2020-03-26 by zssherman