ARM-DOE/pyart #390: Py-ART ideas from the 2015 AMS radar conference

jjhelmus, 2015-10-01:
This is a meta-issue for keeping track of ideas that were discussed with Py-ART users and supporters at the 2015 AMS radar conference.  Expect this top level comment to be edited to reflect changes as development progressed.  Some of these issue already have their own issue in the tracker, I'll add links to these when possible.  

Please provide any additions to this list.  Feel free to share opinions as to the priority of any of these ideas.

### Idea for Py-ART from the conference broken down by area

#### pyart.io

* [x] UF writing (#186, completed in #388) 
* [ ] Reading NEXRAD Level II message 1 format (#352)
* [ ] Reading single field/sweep from NEXRAD Level II
* [ ] Interpolating NEXRAD data to a uniform grid spacing when not-constant (#152, #252)
* [ ] Full support for reading ODIM_H5 files (#131)
* [ ] Writing ODIM_H5 files.

#### pyart.correct

* [ ] Correct unfolded velocities in the region based dealiasing algorithm using a sonde profile or previously unfolded field.
* [ ] Updated phase processing code using self consistency or other additional constraints (#125)

#### pyart.graph
* [ ] Default colormaps based on field name.
* [ ] Default luminosity limits based on field name.
* [ ] Categorical colorbars.
* [ ] Expose calculation of inputs to pcolormesh in the  public API.

#### pyart.grid
* [ ] Three dimensional linear interpolation based gridding as used by SPRINT.
* [ ] Nearest neighbor gridding (#182)
* [ ] Distance weighted gridding where the weighting distance in x, y and z can be specified independently 

#### pyart.retrieve
* [ ] QPE equaltion, R(A), R(KDP), etc.
* [ ] VAD, VVT and DVAD retrievals.  Some of these are available in [SingleDop](https://github.com/nasa/SingleDop)

#### Other 
* [ ] Quasi vertical profiles, QVP.
* [ ] Object for iteration over radar objects by specifying a list of files (useful for QVP).
* [X] Cross section tool for making building RHI like scans from PPI data (completed in #380)
* [ ] Radar composibility (add, remove, insert sweeps or rays) (related #127 and #242)
* [ ] Create conda packages for other radar tools (RSL, wradlib, RadX, PyTDA, SingleDop, ARTView, etc).
* [ ] GridFilter class.
* [ ] GateFilter method to include/exclude based upon height of gate.

#### Documenation
* [ ] Add links to short course materials.

jjhelmus, 2015-10-01:
Should mention that this list is in no way a indication that all of these ideas will be implemented.  Rather that they are being considered.

gamaanderson, 2015-10-01:
I can't find this anymore, but in some issue it was discussed about Grids with non-uniform gates. You may want to add this as a topic there, I think it is worth considering.

scollis, 2015-10-01:
Do you mean radars or grid objects?

gamaanderson, 2015-10-01:
I mean grid objects, i.e. non uniform "x_disp", "y_disp" and "z_disp". 

Radar objects being based in CfRadial should already accept non-uniform range and time axes; there may be some function that presuppose uniformity in Radar, but I would classify that (if not documented) as a Bug.

scollis, 2015-10-01:
Yes. This is work that WILL be done. Either part of Py-ART or something 
compatible with Py-ART. I am about to start a new project that will work 
with grid data in a variety of coordinate systems and attempt to make 
them inter-compatible. eg take a lat lon grid in sigma (terrain 
following) coordinates and easily project to x/y in height
etc...

gamaanderson, 2015-10-01:
Starting a new progress is also good, them we can make a fresh start and extend py-ART later. Of course you are following issues #285 and #294, there are important considerations there.

jjhelmus, 2015-10-01:
Issue #350 covers non-uniform grids.  I'll add a link to the top comment.

jjhelmus, 2016-01-18:
Closing a few NEXRAD related ideas with the merging of PR #455

jjhelmus, 2016-01-28:
> Expose calculation of inputs to pcolormesh in the public API.

Once PR #463 is merged this will be possible using the `get_field` and `get_gate_x_y_z` method of the **Radar** class.  For example to plot a PPI:

``` Python
import matplotlib.pyplot as plt
import pyart

radar = pyart.io.read('XSW110520105408.RAW7HHF')
data = radar.get_field(0, 'reflectivity')
x, y, z = radar.get_gate_x_y_z(0, edges=True)
plt.pcolormesh(x, y, data, vmin=-32, vmax=64.)
plt.show()
```

jjhelmus, 2016-02-02:
With PR #468 merged it is possible to create categorical or non-linear colorbars by specifying the **norm** keyword.  For example:

``` Python
import matplotlib as mpl
import matplotlib.pyplot as plt
import pyart

radar = pyart.io.read('KLOT20130418_043539_V06.gz')

bounds = [-7.5, 12.25, 22.5, 35.5, 77.5]
cmap = mpl.cm.get_cmap('pyart_NWSRef')
norm = mpl.colors.BoundaryNorm(bounds, cmap.N)
display = pyart.graph.RadarDisplay(radar)
display.plot('reflectivity', norm=norm, cmap=cmap)
plt.show()
```

[image]

nguy, 2016-02-03:
This is great! I just implemented something similar in AWOT using a 'discrete_cmap_levels' keyword.

jjhelmus, 2016-03-17:
Updated list to indicate that field specific colormaps and luminosity limits were added by PR #464