ARM-DOE/pyart #415: Issue with reference manual

jwilk88, 2015-10-29:
It appears changes were made in the coding for your user manual recently. I've been using it to learn the ins and outs of RadarMapDisplay, however the RadarMapDisplay reference page seems to have been crossed up with the source code to decorators.py

"Actual wrapper around the function/method." this is all you get in the reference page now (as well as the descriptor for RadarMapDisplay on the Graphing page one directory level up). And the source link takes you to the source code page for decorators.py instead of radarmapdisplay.py.

jjhelmus, 2015-10-29:
Thanks for the report @jwilk88. This issue has been reported in #414 and a fix will be submitted shortly.

jjhelmus, 2015-10-29:
For the time being the [documentation for the latest release (v1.5)](http://arm-doe.github.io/pyart/dev/index.html) should give you documentation on RadarMapDisplay.

jjhelmus, 2015-10-29:
The docs have been rebuild after the merge of #416 and now should be correct. Thanks again for bringing this to our attention.

closed 2015-10-29 by jjhelmus