ARM-DOE/pyart #418: Variable Dependent Nyqvel

gamaanderson, 2015-11-02:
In the cfradial convention nyquist_velocity is global, however when working with unfolded fields it is possible to have different nyquist velocity, which is really important for choosing for instance the colormap. I would like to solve this by adding a variable attribute to overwrite the global nyquist, maintaining the global value as the nyquist velocity of the original measure. Alternately we could also use the CF attribute `valid_range`, at least in my use case (defining a colormap) this would work.  suggestions?

When this is defined, `DEFAULT_FIELD_LIMITS` should be updated accordingly.

jjhelmus, 2015-11-03:
Thanks for the suggestions @gamaanderson.

Just to make things clear, the Nyquist velocity cannot "change", once a spectrum is collected and moment calculated the Nyquist velocity is a set property for the particular gates (and typically ray and sweep).  What can change is what are good limited when plotting the Doppler velocities.  Raw velocities are limited by +/- the Nyquist velocity so these values are good limits when plotting.  If the velocities have been unfolded (dealiased) then these limits are no longer appropriate for plotting as the velocities will be outside of these limits if any folded gates were present in the volume.  Also, it should be noted that it is not uncommon for the Nyquist velocities reported in files to be incorrect, but there is little Py-ART can (and perhaps should) do in these cases.

That said, a method for determining reasonable default limits for unfolded Nyquist velocities would be a good addition to Py-ART.  I would be in favor of updating the function used for determining these limits for the `corrected_velocity` field most likely not the `velocity` field.  How these limits should be determined needs some discussion. Perhaps finding the integer multiple of the Nyquist velocity for which all velocities in the field are inside these limits would be a reasonable default.  Another option would be for the dealiasing routines in Py-ART to set these limits as `valid_min` and `valid_max` attributes in the field dictionary and then add logic to the `get_field_limits` to look for and use these values.  The `parse_vmin_vmax` function in pyart/graph/common already this logic for use in the **RadarDisplay** and related classes.

gamaanderson, 2015-11-03:
Thank you for clarifying @jjhelmus. I'm in favor of using `valid_min` and `valid_min`, first because it is CF conform, and second because it is a general, so may be it is also for non-velocity fields useful.

jjhelmus, 2015-11-03:
Sounds like an idea.  I'm going to be refactoring the **Radar** and **Grid** classes in the next week days to weeks to add in proper lat/lon/alt and x, y, z support (see #247 among others) and I'l guessing some updates and refactoring of the **Display** classes will also be needed.  I can add this feature in during that work since it should fit nicely.  Also been meaning to connect the default colormaps and limits setting from the config file into the **Display** classes.

jjhelmus, 2016-02-02:
Connecting the default colormaps and luminosity limits to the plotting function is implemented in PR#464.

closed 2016-02-02 by jjhelmus

gamaanderson, 2016-02-03:
Thank you for tackling this @jjhelmus!

commit 0eef2de6 (2016-02-02): ENH: Dealiasing routines set valid limits

The dealias_fourdd, dealias_unwrap_phase, dealias_region_based function set the
valid_min and valid_max keys in the returned unwrapped velocity field
dictionary.  This behavior can be changed using the `set_limits` parameter of
this function.

closes #418