ARM-DOE/pyart #421: BUG in _rsl_interface.pyx?

oue, 2015-11-04:
In line 177 of _rsl_interface.pyx, which reads

```python
gate_size = int(radar.range['meters_between_gates'])
```
I'm getting a ValueError because `radar.range['meters_between_gates']` is a unicode, when I am reading Oliktok Ka-SACR PPI data.

jjhelmus, 2015-11-04:
@oue can you provide some additional details on this issue.  Namely what file is causing the issue, what python, pyart, and rsl version are you using, and a full traceback.  All of these will help debug this issue.

oue, 2015-11-04:
Hi @jjhelmus, when I do `print pyart.__version__` I get 1.6.0.dev+6498bb7. I am running Python 2.7. You can download the file I used from https://engineering.arm.gov/~eluke/oli_cga/olikasacrppivhD1.a1/. The specific radar volume was olikasacrhsrhiD1.a1.20151017.121108.nc.

josephhardinee, 2015-11-04:
It appears this is a bug in pyart. In particular, the gate_size is supposed
to be a float value. I'll work on submitting a patch in a little bit. With
that having been said, you can just use pyart.io.read() on those files as
they should be proper CF/Radial files. If they are not, let me know so that
the ingest can be updated.

jjhelmus, 2015-11-04:
Line 177 is in the `label_volume` function which is only invokes when the FourDD dealiasing is used, so reading should not be an issue. The function does the translation from **Radar** instances to Rsl objects. 

I have meeting most of the day but will try to look into this when I have some free time.

jjhelmus, 2015-11-04:
@oue Are you trying to use the `dealias_fourdd` function on this file?  

Py-ART does not seem to have an issue reading the visualizing this file but it does run into some issue if you try to unwrap the Doppler velocities using this function.  Some of these issues are caused by these files not following the CfRadial conventions for file layout.  These include using text and not a float to specify the `meters_between_gates` and `meters_to_center_of_first_gate` attributes of the range variable.  Also many of the instrument parameters variable, including the nyquist_velocity, are scalar where-as CfRadial requires that these have dimensional of time.  

Even fixing these the `dealias_fourdd` function will not work at the moment as it currently does not play nicely with volumes where one or more rays are not included in a sweep (some rays are "in transition").

oue, 2015-11-04:
Hi @jjhelmus : Yes I tried the dealias_fourdd function. I am using a different function (e.g., dealias_region_based), instead. It seems to work well. Thank you.

jjhelmus, 2015-11-05:
@oue Thanks for bringing this to our attention.  I was nearly able to get the `dealias_fourdd` function to work on this radar by adjust the **Radar** instance after reading the data from the file as follows:

``` Python

# read in the radar file
radar = pyart.io.read('olikasacrppivhD1.a1.20151017.120738.nc')

# Fix some issues with the layout
radar.range['meters_between_gates'] = (
    float(radar.range['meters_between_gates']))
radar.range['meters_to_center_of_first_gate'] = (
    float(radar.range['meters_to_center_of_first_gate']))
radar.instrument_parameters['nyquist_velocity']['data'] = (
    np.ones(radar.nrays) *
    radar.instrument_parameters['nyquist_velocity']['data'][0])
```

With these changes I get the following error when trying to call the `dealias_fourdd` function:

```
$ python bug.py 
Traceback (most recent call last):
  File "bug.py", line 38, in <module>
    radar.add_field('corrected_velocity', dealias_data)
  File "/home/jhelmus/dev/pyart/pyart/core/radar.py", line 567, in add_field
    raise ValueError(err)
ValueError: 'data' has invalid shape, should be (3908, 671)
```

This is caused by the inability of the current implementation to deal with radar volumes where not all the rays are included in the defined sweeps (one or more rays are 'in-transition').  I'll one an issue for this bug so that is does not get lost.

jjhelmus, 2015-11-05:
Since #423 describes the bug in more detail I am going to close this issue.  

Feel free to reopen if you have additional problems working with the OLI SACR files in Py-ART.

closed 2015-11-05 by jjhelmus