ARM-DOE/pyart #424: Bug in add_field() grid object

bdolan44, 2015-11-05:
In 'grid.py', 'add_field' function:

Line 79         nz, ny, nx = self.fields[self.fields.keys[0]]['data'].shape

Is missing () after keys such that it should read:

Line 79         nz, ny, nx = self.fields[self.fields.keys()[0]]['data'].shape

(I was getting error: builtin_function_or_method' object has no attribute '__getitem__')

Yay! I have contributed to PyArt!

kirknorth, 2015-11-05:
Hi @bdolan44, this issue was actually recently fixed in PR #409, which means you may want to fetch the latest set of commits to Py-ART's master branch.

bdolan44, 2015-11-05:
Whoops, sorry about that. I'll grab the latest! Thanks!

closed 2015-11-05 by bdolan44

kirknorth, 2015-11-05:
No worries, it still counts as contributing! :smile:

scollis, 2015-11-05:
Of course it does! Thanks Brenda!