ARM-DOE/pyart #428: Masking reflectivity values for attenuation calculations has no effect

habipyesilyurt, 2015-11-13:
In attenuation.py reflectivity values are masked by "refl = np.ma.masked_where(mask, reflectivity_horizontal + z_offset) " expression. But mask is removed after "sm_refl = phase_proc.smooth_and_trim(ray_init_refl, window_len=5)" expression. Then all calculations are done using unmasked reflectivity "sm_refl ". Is that the intent? Or should masked range cells be filled using interpolation?

jjhelmus, 2015-11-16:
I'm not sure what the proper behavior is in this function.  Since attenuation is cumulatinve, the specific attenuation likely needs to be calculated at every gate even if the underlying data is "bad" and should be ignored.  It looks like the masking is applied to the corrected reflectivity field, perhaps this mask should also apply to the specific attenuation?  

Regardless, this function should be updated to use a gatefilter to specify valid gates rather than using the field based minimums.

jjhelmus, 2016-02-08:
Was this closed on purpose?  I do not believe this issue has been fixed.

habipyesilyurt, 2016-02-09:
Sorry!

zssherman, 2022-11-07:
gatefilter exists for zhiphi and philinear, still needs top be added to attenuation function

closed 2022-11-09 by zssherman