ARM-DOE/pyart #432: Auxillary Data

josephhardinee, 2015-11-24:
I figure I'd start this discussion here. I'm planning on adapting the newer SACR reader to put it into pyart. The reasons for this issue however is that there is a large amount of auxillary data in these files that do not have a clear representation in the current radar object. While they could be made fields in the field dictionary, they are primarily metadata, and there is a large enough amount of them that I don't want to clutter up the radar object fields or info unnecessarily. 
So the question is, what is the best way to incorporate these into the radar object. I was thinking something like an aux_info dictionary being added might be the best way. This would be done by the reader after creation and so would not need any changes to pyart. These are time series so they wouldn't need pyart machinery to plot them. 
What do you think?

jjhelmus, 2015-11-24:
Could such information be put into the *metadata* dictionary?  

The *metadata* dictionary is used as a catch all for information that does not map nicely from the Cf/Radial conventions to the **Radar** class and I don't think we need to add yet another "catch all" attribute off the class which is already getting quite bulky.  Some [information is taken from this dictionary](https://github.com/ARM-DOE/pyart/blob/de9faa499c28ed243220477a496660698899b467/pyart/io/cfradial.py#L451-L481) when calling the `write_cfradial` function but I think this can be dealt with if issues arise. The **Radar** class does not perform any checking of the layout or contents of the dictionary so it should be safe organize radar specific information from the SACR or other instrument in this dictionary. 

Just a heads up, I do not want Py-ART to start interpreting and interrogating any such ad-hoc "auxillary" information to make decision inside algorithms or routines.  This very quickly leads to a loss of generality and complications.

josephhardinee, 2015-11-24:
Sure it can go in there, I wasn't sure if any of the core functions relied on metadata not having extra fields. None of these fields should overlap with the fields that already exist in there. 

The plan is that this data will not be important for any of pyart's algorithms(It includes things like voltages, fault states, other things that are hyper radar specific). There is a lot of extra data in these files, and if users are going back to the .00 files, they might want those parameters that we strip out in the .a1 files. Your comment answers my questions though so I'll go ahead and close this.

closed 2015-11-24 by josephhardinee