ARM-DOE/pyart #440: pyart.io.common.radar_coords_to_cart in Py-ART version 1.6

tanelv, 2015-12-14:
Having upgraded from version 1.0 to 1.6 I can not see radar_coords_to_cart function in pyart.io.common module.
Listing the available functions only gives these:
>>> dir(pyart.io.common)
['__builtins__', '__doc__', '__file__', '__name__', '__package__', '_test_arguments', 'bz2', 'gzip', 'make_time_unit_str', 'netCDF4', 'np', 'prepare_for_read', 'stringarray_to_chararray']

I saw some warnings in the update, but no errors, so the upgrade seemed to be OK.

What could be the case?

tanelv, 2015-12-14:
Already found a workaround!
I checked the common.py file in the /io directory and there was no radar_coords_to_cart function defined. But looking at the older common.py file from the previous pyart version installation (I made a copy of the files pre installation of the new version) the function was there. So I copied the function to the new common.py and now it works.
There was a comment in the old common.py file above the radar_coords_to_cart function:
# XXX move this to another module

So it should be in some other module?

def radar_coords_to_cart(rng, az, ele, debug=False):
    """
    Calculate Cartesian coordinate from radar coordinates

    Parameters
    ----------
    rng : array
        Distances to the center of the radar gates (bins) in kilometers.
    az : array
        Azimuth angle of the radar in degrees.
    ele : array
        Elevation angle of the radar in degrees.

    Returns
    -------
    x, y, z : array
        Cartesian coordinates in meters from the radar.

    Notes
    -----
    The calculation for Cartesian coordinate is adapted from equations
    2.28(b) and 2.28(c) of Doviak and Zrnic [1]_ assuming a
    standard atmosphere (4/3 Earth's radius model).

    .. math::

        z = \\sqrt{r^2+R^2+r*R*sin(\\theta_e)} - R

        s = R * arcsin(\\frac{r*cos(\\theta_e)}{R+z})

        x = s * sin(\\theta_a)

        y = s * cos(\\theta_a)

    Where r is the distance from the radar to the center of the gate,
    :math:`\\theta_a` is the azimuth angle, :math:`\\theta_e` is the
    elevation angle, s is the arc length, and R is the effective radius
    of the earth, taken to be 4/3 the mean radius of earth (6371 km).

    References
    ----------
    .. [1] Doviak and Zrnic, Doppler Radar and Weather Observations, Second
        Edition, 1993, p. 21.

    """
    theta_e = ele * np.pi / 180.0       # elevation angle in radians.
    theta_a = az * np.pi / 180.0        # azimuth angle in radians.
    R = 6371.0 * 1000.0 * 4.0 / 3.0     # effective radius of earth in meters.
    r = rng * 1000.0                    # distances to gates in meters.

    z = (r ** 2 + R ** 2 + 2.0 * r * R * np.sin(theta_e)) ** 0.5 - R
    s = R * np.arcsin(r * np.cos(theta_e) / (R + z))  # arc length in m.
    x = s * np.sin(theta_a)
    y = s * np.cos(theta_a)
    return x, y, z

jjhelmus, 2015-12-14:
The `radar_coords_to_cart` function was recently (PR #433) moved and renamed and can now be found  as `pyart.core.transforms.antenna_to_cartesian`.  If you upgrade to the version of Py-ART which includes the changed merged today (PR #439) then `pyart.io.common.radar_coords_to_cart` will also be available, but will emit a DepreciatedFunctionName warning to inform you of the change in name.  

As a heads-up, Py-ART 1.6.0 has not been release.  You are welcome to use the current development version, 1.6.0+dev, but be aware that this version is not as polished as a formal release.  There may be open bugs or incomplete behaviors that are actively being developed.  

Also, I should point out that function like `radar_coords_to_cart` that are only exposed in modules (pyart.subpackage.module.function) and not directly in a subpackage (pyart.subpackage.function) are not considered to be in the public API.  This means that the Py-ART developers expect that changes to the name or behavior of these functions should not effect uses. In turn users should not be surprised if these functions change.  The names and behavior of the functions and classes in the public API, which are documented in the User Reference Manual [http://arm-doe.github.io/pyart-docs-travis/user_reference/io.html], are not expected to significantly change without going through a depreciation cycle in multiple released version of Py-ART.

jjhelmus, 2015-12-14:
Closing this issue as I believe #439 should make these functions available again.  

@tanelv, thanks for reporting this and feel free to re-open this issue if you do not feel as if it has been resolved.

closed 2015-12-14 by jjhelmus