ARM-DOE/pyart #442: The toa parameter is interpreted differently in the two gridding methods

jjhelmus, 2015-12-14:
The `toa` parameter control the height above which gates are excluded from the gridding algorithm.  This parameter is interpreted differently by the two gridding routines.  

In the 'map_gates_to_grid' the distance of the gate above the radar is added to the offset of the radar from the grid and compared against toa (toa is the height above grid origin).  

In 'map_to_grid' the distance of the gates above the radar is compared directly to toa (toa is the height above the radar).  

A uniform interpretation of toa should be used in both gridding methods.

gamaanderson, 2015-12-21:
So we have three options,

1. height above origin
2. height above radar
3. height above see level

I'm for the third one!

jjhelmus, 2016-01-04:
Another option is to remove to `toa` parameter entirely and have these gates removed by the gate filter.  Adding `{include/exclude}_{above/below}_{height/altitude}` methods to the GateFilter for including/excluding gates based upon their height above the radar or altitude should make possible.

scollis, 2016-01-04:
Exactly.. TOA was introduced to speed gridding by rejecting gates that 
will never have returns in them.. Gatefilter effectively handles this now.

PR #1586 (2024-06-20): ADD: Addition of above_toa_filter.
This filter replaces toa parameter in gridding code and allows for the user to specify their own remove above toa filter by using the gatefilter for consistency between both gridding functions.

- [x] Closes #442 
- [x] Documentation reflects changes
closed 2024-06-20 by zssherman