ARM-DOE/pyart #445: odimh5 support for RHI, PPI and sector PPI objects

joshuass, 2015-12-17:
_pyART only appears to support PVOL objects in odimh5_
Single PPI scans and sector scans are defined as SCAN objects
RHI scans are defined as ELEV objects

See page 10 of the specifications doc for more information
http://www.eumetnet.eu/sites/default/files/OPERA2014_O4_ODIM_H5-v2.2.pdf

jjhelmus, 2015-12-17:
@joshuass Can you provide example files of any of these scan modes?  Implementing support for the reading of these files is much easier with examples to work from.

swnesbitt, 2015-12-17:
I have a functional code that can do this (at least the output from BALTRAD
bufr2hdf5 with an Argentinean radar file), just haven't uploaded it to the
archive.

It is in my fork of pyART.

-Steve

joshuass, 2015-12-18:
Hi,

Here is an example RHI file
https://dl.dropboxusercontent.com/u/2037428/test_odimh5_files/uq-xpol_rhi_20151104_184723.h5
Here is an example SPPI file
https://dl.dropboxusercontent.com/u/2037428/test_odimh5_files/uq-xpol_sppi_20151210_162502.h5

I can get the sector PPI files working in pyart by changing the object type to PVOL and setting the azimuth manually. It appears pyART isn't reading the azstart and azstop values for sector PPIs.

Cheers,

jjhelmus, 2015-12-18:
Thanks for the example files @joshuass.  I'll look into adding support to Py-ART for reading in these types of files.

joshuass, 2016-06-09:
Can confirm RHI and PPI odimh5 files are working fine are working perfectly with the new odim reader. You can close this issue and #484 . Thanks for getting it to this point! :+1:

jjhelmus, 2016-06-13:
Thanks for the conformation @joshuass, closing.

closed 2016-06-13 by jjhelmus