ARM-DOE/pyart #448: Transform a nc file data in grid data.

candidobugarin, 2016-01-06:
Hello. I need get the data inside a nc file and put this data in a grid to get a values in some points. Is it possible do that with ART?

Thanks

jjhelmus, 2016-01-06:
What is the source of this NetCDF file?  Does it contain radar data?  Does the layout of the file follow the [Cf/Radial conventions](https://www.ral.ucar.edu/projects/titan/docs/radial_formats/cfradial.html) or another one of Py-ART [supported formats](http://arm-doe.github.io/pyart-docs-travis/user_reference/io.html)?  If so then it should be possible to read the file with Py-ART and map the radial points onto a Cartesian grid.  

Typically question of this type are better asked and answered on the [Py-ART mailing list](http://groups.google.com/group/pyart-users/) if you want to join.

candidobugarin, 2016-01-07:
Hello.
Yes, nc contains radar data. I can any file format, if you tell me the
best format to do this, I'll be happy.
My idea is to tell the latitude / longitude to an array and the array
would return me a DBZ value or rain.
And will be , a pleasure to join you in the mailing list.
Thanks

jjhelmus, 2016-01-18:
The following script will print out the reflectivity of the nearest radar gate and the interpolated reflectivity to a given location specified in the script.  I believe this can be adapted for your use @candidobugarin.  

``` Python
from __future__ import print_function
import numpy as np
import pyart
import pyproj

def find_x_y_displacement(radar, longitude, latitude):
    """ Return the x and y displacement (in meters) from a radar location. """
    # longitude and latitude in degrees
    lat_0 = radar.latitude['data'][0]
    lon_0 = radar.longitude['data'][0]
    proj = pyproj.Proj(proj='aeqd', lon_0=lon_0, lat_0=lat_0)
    return proj(longitude, latitude)

def find_nearest_gate(radar, longitude, latitude, altitude):
    """ Return the indices of the nearest gate to a given point. """
    # longitude and latitude in degrees, altitude in meters
    gate_x = radar.gate_x['data']
    gate_y = radar.gate_y['data']
    gate_z = radar.gate_z['data']

    x_disp, y_disp = find_x_y_displacement(radar, longitude, latitude)
    distances = np.sqrt(
        (gate_x-x_disp)**2. + (gate_y-y_disp)**2. + (gate_z-altitude)**2.)
    return np.unravel_index(distances.argmin(), distances.shape)

def interpolate_single_point(radar, longitude, latitude, altitude):
    """ Interpolate a single grid point at a given location. """
    x_disp, y_disp = find_x_y_displacement(radar, longitude, latitude)
    grid = pyart.map.grid_from_radars(
        (radar,),
        gridding_algo='map_gates_to_grid',
        grid_shape=(1, 1, 1),
        grid_limits=((altitude, 0), (y_disp, 25000.0), (x_disp, 25000.0)),
        fields=['reflectivity'])
    return grid

# read in the file
filename = 'radar_file.nc'
radar = pyart.io.read(filename)

latitude = 80.0        # latitude (in degrees) to find reflectivity at or near
longitude = 120.0   # longitude (in degrees) to find reflectivity at or near
altitude = 1000       # altitude (in meters) to find reflectivity at or near

# find nearest gate
ray, gate = find_nearest_gate(radar, longitude, latitude, altitude)
gate_latitude = radar.gate_latitude['data'][ray, gate]
gate_longitude = radar.gate_longitude['data'][ray, gate]
gate_altitude = radar.gate_altitude['data'][ray, gate]
gate_reflectivity = radar.fields['reflectivity']['data'][ray, gate]
print("-----------------------")
print("Nearest gate:")
print("Latitude:", gate_latitude)
print("Longitude:", gate_longitude)
print("Altitude:", gate_altitude)
print("Reflectivity:", gate_reflectivity)

# interpolate around gate
grid = interpolate_single_point(radar, longitude, latitude, altitude)
pyart.io.add_2d_latlon_axis(grid)
pixel_latitude = grid.axes['latitude']['data'][0, 0]
pixel_longitude = grid.axes['longitude']['data'][0, 0]
pixel_altitude = grid.axes['z_disp']['data'][0]
pixel_reflectivity = grid.fields['reflectivity']['data'][0, 0]

print("-----------------------")
print("Interpolated pixel:")
print("Latitude:", pixel_latitude)
print("Longitude:", pixel_longitude)
print("Altitude:", pixel_altitude)
print("Reflectivity:", pixel_reflectivity)
```

A similar thread on this topic was recently posted to the email list.

candidobugarin, 2016-01-18:
Hello! Thanks for the reply , but over time they expected their response I
looked at your list in gmail and found the answer to the problem I had.
Thanks a lot for your help . I am sending him a link for you to see what I
did with the data. I made the DBZ column for each latitude and longitude
for each city in the state of Rio de Janeiro and found the highest rainfall
for each city and shown on a map. For each color on the map there is a
level of rain. This is the link:
http://alertadecheias.com.br/chuva/index_s.html Any questions, can talk to
me , I want to contribute you.

jjhelmus, 2016-01-19:
Glad that you found a solution to your problem.  Your example looks very interesting. I as well as others Py-ART mailing list would be very interesting is seeing how this was accomplished if you are willing to share the code.  

I'm going to close this issue as you seem to have solved your initial question.  If you are interested in contributing to Py-ART, I would encourage you to post to the mailing list.

closed 2016-01-19 by jjhelmus