ARM-DOE/pyart #450: Need better canned examples plotting data on top of radar data

scollis, 2016-01-08:
Talking to Josh from UQ who was asking about plotting contours on radar map displays. We can do this but we need better documentation and canned examples. 

We do have this example which needs updating:
http://arm-doe.github.io/pyart/dev/auto_examples/plotting/plot_grid_three_panel.html

We need an example plotting onto a PPI via radarmapdisplay. I suggest we use something like WRF data for this one, but I am open to suggestions. 

We need cookbook like examples people can easily take and modify.

scollis, 2016-01-08:
Ping @joshuass

scollis, 2016-01-29:
Ping @Rumpkie : How goes it?

Rumpkie, 2016-01-29:
It is going well; I have a clean simple version of the plot_grid_three_panel.py.  It has the one grid plot and base map with code comments.  I am working on a brief instructional paragraph now for the documentation.

joshuass, 2016-01-29:
Sorry, been away on holidays! Here is an example of the plotting capability I was suggesting...
[image]
It's a true RHI from CP2 showing colour shaded Doppler velocity and unfilled contoured reflectivity in 5dBZ steps from 35dBZ. I've produced similar plots for radar PPIs and Doppler lidar data too. It'll also be very useful for dual-pol analysis.

scollis, 2016-01-29:
Excellent. So not even overlaying from another source.. juts a contour of one moment over a colormesh from another.. 
Could you share the CP2 file?

joshuass, 2016-02-04:
Here are a couple of mdv rhi files from CP2 during my field campaign
https://dl.dropboxusercontent.com/u/2037428/pyart_rhi_testing/034142.mdv
https://dl.dropboxusercontent.com/u/2037428/pyart_rhi_testing/034215.mdv

scollis, 2016-02-04:
Thanks @joshuass. @Rumpkie Could you please download and try to start reproducing the example @joshuass produced. When you get stuck talk to @jjhelmus

Rumpkie, 2016-02-04:
Sounds good @scollis, I will get started on that today.

Rumpkie, 2016-02-08:
I'm getting close to @joshuass's example:

[image]

with the contour lines level set to 8 it gets a lot closer to the figure.
[image]

Here are the contours over their data.
[image]

joshuass, 2016-02-09:
Great work, looks pretty much ready to go! I forgot to mention I used a smoothing filter, but that's something anyone can add on.

scollis, 2016-02-09:
Very nice.. We need some labeling or some coloring and a legend for the lines.. Great progress

Rumpkie, 2016-02-09:
How about this?
[image]

Rumpkie, 2016-02-09:
Here is what I have if anybody wants to play with it so far.
```
"""
====================================================================
Create an RHI plot with reflectivity contour lines from an MDV file
====================================================================

An example which creates an RHI plot of velocity using a RadarDisplay object
and adding Reflectivity contours from the same MDV file.

"""
print __doc__

# Author: Cory Weber (cweber@anl.gov)
# License: BSD 3 clause
import matplotlib.pyplot as plt
import pyart
import numpy as np
import scipy.ndimage as spyi

filename = 'data/034142.mdv'

# create the plot using RadarDisplay
sweep = 2
# read file
radar = pyart.io.read_mdv(filename)
display = pyart.graph.RadarDisplay(radar)
fig = plt.figure(figsize=[20, 5])
ax = fig.add_subplot(111)

# plot velocity
display.plot('velocity', sweep=sweep, vmin=-20, vmax=20.0, fig=fig,
             ax=ax, cmap='pyart_BuDRd18', colorbar_label='Velocity (m/s)')

# line commented out to show reflectivity
# display.plot('reflectivity', sweep=sweep, vmin=-0, vmax=45.0, fig=fig,ax=ax)

# get data
start = radar.get_start(sweep)
end = radar.get_end(sweep) + 1
data = radar.fields['reflectivity']['data'][start:end]

x = radar.gate_x['data'][start:end] / 1000.0
y = radar.gate_y['data'][start:end] / 1000.0
z = radar.gate_z['data'][start:end] / 1000.0

# smooth out the lines
data = spyi.gaussian_filter(data, sigma=1.2)

# calculate (R)ange
R = np.sqrt(x ** 2 + y ** 2) * np.sign(y)
R = -R
display.set_limits(xlim=[25, 0], ylim=[0, 5])

# add contours
# creates steps 35 to 100 by 5
levels = np.arange(35, 100, 5)
# adds coutours to plot
contours = ax.contour(R, z, data, levels, linewidths=1.5, colors='k',
                      linestyles='solid', antialiased=True)

# adds contour labels (fmt= '%r' displays 10.0 vs 10.0000)
plt.clabel(contours, levels, fmt='%r', inline=True, fontsize=10)

# format plot
# add grid (dotted lines, major axis only)
ax.grid(color='k', linestyle=':', linewidth=1, which='major')

# horizontal
ax.axhline(0.9, 0, 1, linestyle='solid', color='k', linewidth=2)
ax.axhline(1.3, 0, 1, linestyle='dashed', color='k', linewidth=2)
# vertical
ax.axvline(15, 0, 1, linestyle='solid', color='#00b4ff', linewidth=2)
ax.axvline(4.5, 0, 1, linestyle='solid', color='#ff6800', linewidth=2)

ax.set_ylabel('Altitude above CP-2 (km)')

plt.show()

```

joshuass, 2016-02-10:
Looks perfect! I've saved this script for my own future reference but it would be great to include as part of the py-art tutorial. Feel free to include the mdv I provided in the tutorial too.

It'll be a fantastic tool for integrating the analysis of dual pol moments too!

jjhelmus, 2016-02-10:
Looking really good @Rumpkie.  Just two suggestions:

* scipy.ndimage is typically imported as ndimage.

* Use the **Radar** methods to retrieve the field data and gate locations rather then slicing the data and worrying about getting start and end correct.  The following will work:

``` Python
data = radar.get_field(sweep, 'reflectivity')
x, y, z = radar.get_gate_x_y_z(sweep, edges=False)
x /= 1000.   # meters to kilometers
y /= 1000.
z /= 1000.
```

These methods were recent additions to Py-ART so you may need to update you repository.

* The ylabel could be set using the _axislabels_ parameter in the `display.plot` method, but I actually like the current way this is done, it shows that matplotlib can be used to override the defaults.

Rumpkie, 2016-02-18:
@scollis , here is what I have so far.

[image]

scollis, 2016-02-18:
Very nice.. Question is how do we make the significant contour lines stand out.. @joshuass @josephhardinee @jjhelmus ? There is a nice region of big drops around the core.. And an area of negative ZDR aloft with vertically aligned ice..

Rumpkie, 2016-02-18:
I wouldn't, I would leave the contour lines we have(perhaps remove a few steps to simplify it), 
Filter the data to isolate large rain drops into a binary grid and add contours from with text labels.
Repeat that process for one that isolates vertically aligned ice.

The question is if matplotlib can handle multiple contour designations.... I think it one of the examples it does, I'll go digging tomorrow.

joshuass, 2016-02-18:
I've seen bolder contours in larger steps (say 1.5dB in this case). That
might be worth trying. Otherwise it looks amazing. Looking forward to
trying it on Doppler scanning lidar data too.

swnesbitt, 2016-02-18:
ahh, NPOL was down during this time period (2142-2321)...

Rumpkie, 2016-02-18:
I spent a bit more time on cleaning it up... I'm not sure if this is what @scollis had in mind but:

[image]

josephhardinee, 2016-02-19:
The contours look nice now, but the reflectivity appears fairly washed out.

scollis, 2016-02-19:
Agree @Rumpkie .. No Alpha..

josephhardinee, 2016-02-19:
Though I will say that the contours are looking quite nice.

Rumpkie, 2016-02-20:
@scollis, so without the partial transparency, the lines are much less visually striking, some get lost altogether, the text gets harder to read, and of-course the example looses the valuable code and explanation about how to do the alpha blending.

[image]

Furthermore, if you look at 75% alpha blending, there is a problem with the way matplotlib draws pcolormesh with transparency, it never handles the alpha on edges correctly even if edgecolors are set to 'face' or 'none'. In the previous example, I had mitigated the issue by tweaking the alpha of the lines themselves, but I can't seem to get it to play nice at 75% no matter what I do.

notice the Moiré pattern:
[image]

This is caused by the light colored lines on the edges. They can also be seen clearly in the color bar.
[image]

joshuass, 2016-04-28:
I've spent some time extending this functionality to PPI plots for my phd. It works very well!
https://github.com/joshuass/notebooks/blob/master/pyart/CP2_20141127_obspaper.ipynb

scollis, 2016-09-14:
I think we are done with this particular enhancement.. Closing

closed 2016-09-14 by scollis