ARM-DOE/pyart #451: Time-height plotting

nguy, 2016-01-08:
Currently Py-ART does not support time-height plots it seems. The VPT (vertically-pointing) plot function plots the abscissa as distance ( I assume for a scanning type?). I would suggest modifying the ```RadarDisplay.plot_vpt``` to accept (or access) a time variable and format accordingly. Maybe use of a keyword?

This topic had some discussion in Issue #364 .

One potential issue would be to make sure that time was represented consistently.

nguy, 2016-01-08:
I found some time conversion utilities in ```pyart.util.datetime_utils.py```, but cannot find an instance of these being used in the package. 

I think adding a conversion to Epoch time here would be useful in the future when potentially combining radar instances.

jjhelmus, 2016-01-15:
@nguy Sounds like a great suggestion.  The current VPT plots do not properly represent gaps in time or even indicate the timing of the measurement.  I'll have a look at #453 later today and make some comments/suggestions.

jjhelmus, 2016-01-19:
Closed by PR #453

closed 2016-01-19 by jjhelmus