ARM-DOE/pyart #458: radar.info output

nguy, 2016-01-22:
I tried to save the output to a text file and received the following error below. Is this an issue because I'm using python 2.x and not 3?

guy:20130520 $ python
Python 2.7.11 |Anaconda 2.1.0 (x86_64)| (default, Dec  6 2015, 18:57:58) 
[GCC 4.2.1 (Apple Inc. build 5577)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
Anaconda is brought to you by Continuum Analytics.
Please check out: http://continuum.io/thanks and https://anaconda.org
>>> import pyart
>>> r=pyart.io.read('KTLX20130520_222418_V06')
>>> r.info(out='longradar.txt')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/guy/anaconda/lib/python2.7/site-packages/pyart/core/radar.py", line 525, in info
    self._dic_info('altitude', level, out)
  File "/Users/guy/anaconda/lib/python2.7/site-packages/pyart/core/radar.py", line 615, in _dic_info
    print(ilvl0 + attr + ':', file=out)
AttributeError: 'str' object has no attribute 'write'

jjhelmus, 2016-01-22:
The out parameter to the `Radar.info` function is a file-like object (anything with a `write` method), probably the best example is a file opened for writing.  So you need to do the following:

``` Python
import pyart
radar = pyart.io.read('KTLX20130520_222418_V06')
file_obj = open('longradar.txt', 'w')
radar.info(out=file_obj)
file_obj.close()
```

This is a bit more work that providing a file name but provides additional functionality (printing info on multiple radars to a single file, logging, dumping to a gzip file, standard error, etc).   

We could also have the method accept a string filename which written to and add a bit of logic to do type checking.

nguy, 2016-01-22:
Thanks, I misunderstood the useage there. Works for me now. 
I'd say any enhancement is really low priority and entirely up to you. If you do decide to change, please let me know so I can test with ARTView.

jjhelmus, 2016-01-22:
Good to hear, Python filenames vs file object are confusing, I've been flustered more than once with the difference.  

I'm going to hold off adding support for providing a filename to to info method because the same result can be achieved with one additional line if you use a context manager:

```Python
with open('longradar.txt', 'w') as f:
   radar.info(out=f)
```

closed 2016-01-22 by jjhelmus