ARM-DOE/pyart #461: AWR XSACR VPT files plot incorrectly.

josephhardinee, 2016-01-26:
Just a heads up, the plot_vpt function exhibits some weird behavior on the AWR XSACR VPT files. 
File was in read in with pyart.read, and plotted with display.plot_vpt. This causes only about 35 rays to be plotted, although sweep start index, sweep end index, nrays, and ngates all appear fine. I put the file at 
/home/hardin/awr_ingest/awrxsacrvptM1.a1.20160121.084234.nc
on the DMF. It wasn't immediately obvious what the issue was.

jjhelmus, 2016-01-26:
There are only 32 rays in that volume where the antenna_transition variable indicates that the radar was not transitioning between sweeps.  If you plot using with `filter_transitions=False` all 700+ rays will be included in the plot.  The default is to filter these transitions.

Should be noted that transition filtering does not work if the `time_axis_flag` is to True.

josephhardinee, 2016-01-26:
Okay thanks, I didn't think to check that. Closing the issue.

closed 2016-01-26 by josephhardinee