ARM-DOE/pyart #474: Display method for plotting locations of azimuth angles

jjhelmus, 2016-02-03:
As suggested by @nguy on the Py-ART mailing list a method for plotting a line along a particular azimuthal angle on top of a plot of a PPI would make a good addition to Py-ART.

gamaanderson, 2016-02-03:
ideally such a line should not be a strait line, but slightly curved depending on the map projection.

jjhelmus, 2016-02-03:
Agreed, although only on maps.  I think the line should be straight when generated with **RadarDisplay**. Would transforming the Cartesian coordinates of the line into Geographic (latitude and longitude) points would results in a curved line.

nguy, 2016-02-03:
I think a straigh line is the way to go with **RadarDisplay**. The code in ```pyart.util.xsect.cross_section_ppi``` should do a lot of the work for this, right? The Cartesian and geographic (fed back into the basemap instance) should be able to be used for the line if I'm thinking of this correctly.

jjhelmus, 2016-02-04:
The `antenna_to_cartesian` function will work for the **RadarDisplay** for plotting on maps the best solution is probably to also transforms these to geographic coordinate, I have to think about this a bit more.

jjhelmus, 2016-02-04:
Unless I'm not understanding the use of this function, I don't think `pyart.util.xsect.cross_section_ppi` will be of use here.

nguy, 2016-02-04:
Not the function, I was just floating through some of the code. I was also thinking of having the ability to select a line for cross-sectioning, like can be done [here](http://nmq.ou.edu/applications/qvs_3d_tools_main.html) in the future, which is why I was rummaging through that particular code.

Not sure if this would be a Py-ART or ARTview thing though.

nguy, 2016-02-04:
I should say the GUI part would obviously be ARTview, but where do we put the code under the hood?

jjhelmus, 2016-02-04:
Scott and I were talking a few days ago about adding methods to the **Radar** and **Grid** class which return the indices for the nearest gate or point to a set of coordinates (either x, y, z or longitude, latitude and altitude).  I think with methods like this, lookup of the data would be quite straightforward.  I opened an issue (#476) for this feature.

jlpippitt, 2020-06-11:
I know this is an old post and you may have implemented new code.  I was able to plot azimuths doing the following. 

    dtor = math.pi/180.0
    max_range=150.0
    maxrange_meters = max_range * 1000.
    meters_to_lat = 1. / 111177.
    meters_to_lon =  1. / (111177. * math.cos(radar_lat * dtor))

    for azi in range(0,360,30):
            azimuth = 90. - azi
            dazimuth = azimuth * dtor
            lon_maxrange = radar_lon + math.cos(dazimuth) * meters_to_lon * maxrange_meters
            lat_maxrange = radar_lat + math.sin(dazimuth) * meters_to_lat * maxrange_meters
            display.plot_line_geo([radar_lon, lon_maxrange], [radar_lat, lat_maxrange],line_style='k-',lw=0.5)

[image]

zssherman, 2020-06-11:
Thanks @jlpippitt ! I can't remember if this was ever added, but would be great to add to an example or function if it never was. I'll look into that.