ARM-DOE/pyart #476: Methods for finding the indices of nearest gate/point

jjhelmus, 2016-02-04:
Adding methods to the **Radar** and **Grid** class for finding the indices of the nearest gates or grid point to a set of coordinates would be useful.  Accepting coordinate in either Cartesian (x, y, z) or Geographic (Longitude, Latitude, Altitude) would be recommended, with the height coordinate being optional.  Additionally being able to limit the search to a specific sweep or z-level would be good.

This feature could also be implemented as an independent function and not a class method, likely being placed in the `util.radar_utils` module.  

With this capability it would be possible to extract out the nearest field values for a set of coordinate by using the indices returned by this method and "fancy" array indexing.

nguy, 2016-02-04:
I've been toying with something that is [similar](https://github.com/nguy/AWOT/blob/master/awot/util/matcher.py) where a flight track is matched to for example a ground-based radar. There are a couple of methods developed by @tjlang and George Duffy that I ported and modified. The index retrieval is in the code. There are two nearest neihbor and a KDTree method. 

I don't know if these are helpful, but maybe worth a look?

gamaanderson, 2016-02-04:
+1 for this. We have implemented something similar in artview and would love to use an pyart version.

Something that we could also use is a version of this in the display class, that take as argument the map coordinates (that is from ax), but i don't know if that make much sense in pyart or if we would use it in artview (we may want to maintain control over this kind of calculation)

closed 2022-11-03 by mgrover1