ARM-DOE/pyart #478: depreciation warnings in python 3.5

swnesbitt, 2016-02-04:
Just passing along

```
(py3k)pc105:python snesbitt$ ipython
Python 3.5.0 |Anaconda 2.4.1 (x86_64)| (default, Sep 13 2015, 10:37:42) 
Type "copyright", "credits" or "license" for more information.

IPython 4.1.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import pyart
/Users/snesbitt/anaconda/envs/py3k/lib/python3.5/site-packages/scipy/_lib/decorator.py:205: DeprecationWarning: inspect.getargspec() is deprecated, use inspect.signature() instead
  first = inspect.getargspec(caller)[0][0]  # first arg
/Users/snesbitt/anaconda/envs/py3k/lib/python3.5/site-packages/pyart/map/grid_mapper.py:40: DeprecationWarning: inspect.getargspec() is deprecated, use inspect.signature() instead
  from .ball_tree import BallTree
```

jjhelmus, 2016-02-05:
Looks to be an issue with scipy (or rather the vendorized version of decorators which is included in scipy).  This was  addressed in scipy/scipy#5473 and should be fixed in scipy v0.17.0.  Hopefully conda packages for 0.17.0 will be release soon.

jjhelmus, 2016-02-05:
Thanks for the report, by-the-way, always good to hear from you @swnesbitt .

jjhelmus, 2016-02-05:
Upgrading to SciPy 0.17.0 removed one of the DeprecationWarnings but there is still one being emitted by `from .ball_tree import BallTree`, specifically [line 272 in pyart/map/ball_tree.pyx](https://github.com/ARM-DOE/pyart/blob/master/pyart/map/ball_tree.pyx#L272).  

That code is originally from sklearn, and they have a fix which involves a lot of additional baggage.  I'm tempted to remove the entire module since the BallTree is typically no faster than a Kd-Tree for the uses in Py-ART.  Also, the map_gates_to_grid algorithm typically performs better than map_to_grid no matter what type of tree is used.

PR #504 (2016-03-16): MAINT: Remove BallTree class
The BallTree class is not longer needed as the k-d tree is typically fast
enough and the map_gates_to_grid algorithm is available if faster gridding is
required.

closes #478
commit 96455402 (2016-03-16): MAINT: Remove BallTree class

The BallTree class is not longer needed as the k-d tree is typically fast
enough and the map_gates_to_grid algorithm is available if faster gridding is
required.

closes #478
closed 2016-03-16 by jjhelmus