ARM-DOE/pyart #483: support for cartopy for mapping

swnesbitt, 2016-02-10:
I am increasingly favoring Cartopy as a mapping package over Basemap.  I find it to be much higher performance, has more meteorological community backing, and has lots of built in GIS capabilities, as well as integration with web mapping services like google and NASA.

From an architecture standpoint, I am wondering what the best way to implement this would be.  It would be nice to be able to select the mapping backend in functions like `plot_ppi_map` through a call to a `map_backend` keyword, and then allow a dictionary to be passed that could control the functionality of the mapping, and return handles to the plotting so that further cartopy functionality could be applied (i.e. overlay of shapefiles).

I am willing to take a crack at this, just interested in some general feedback on the idea.

scollis, 2016-02-10:
Completely concur. and yes. I do think having a backend keyword would be the best way to implement this. To start with Matplotlib will be the default but if cartopy becomes more fully featured it can eventually switch. An error should be raised if the requisite package does not exist. ie cartopy will not be a compulsory dependency for Py-ART

jjhelmus, 2016-02-10:
:+1: Having support for plotting on maps with Cartopy would be great.
 
In terms for implementation, if I was going to work on this I would begin by creating a separate class which uses Cartopy for mapping say **RadarMapDisplayCartopy**.  Get that working first and then try to merge the two classes with the backend being an optional argument.  I think this approach would be a bit easier than trying to work out all the details in the **RadarMapDisplay** class from the get go.

scollis, 2016-04-14:
We discussed this at the NCAR SEA.. Our discussion was prompted by MetPy support for Cartopy and station plots.. 
One issue is that Cartopy is not yet in the main channel of Conda. While we have some dependencies that are not in conda they are very optional and in direct support to ARM programs (eg CyLP). 

I will contact the Cartopy leads and check on their plans and if they intend to get a Conda package up.. If the answer is yes then we can think about adding this to the roadmap (increasing chances for DoE support through Argonne). Otherwise this remains in the domain of a user contribution and will need a conditional import.

jjhelmus, 2016-04-14:
There are a number of cartopy conda packages on Anaconda.org, including the ones in the [conda-forge](https://anaconda.org/conda-forge/cartopy) channel which are likely the "best".  The concern is that cartopy is not in the default channel nor installed in the anaconda meta-package.  Both of these are managed by Continuum and getting cartopy included in either would have to have a change on their end.

scollis, 2016-04-14:
So,
would a conda install from the jjhelmus channel resolve dependancies outside of the default channel?

the key test is “will this update make Py-ART harder to install?”

Dr Scott Collis
Purveyor of lofty observations (Atmospheric Scientist) 
Environmental Science Division
Argonne National Laboratory
Mb: +1 630 235 8025
Of: +1 630 252 0550
Become a Py-ART user today! http://arm-doe.github.io/pyart/ <http://arm-doe.github.io/pyart/>

jjhelmus, 2016-04-14:
`conda install -c jjhelmus pyart` will search the `jjhelmus` channel and the default channels for package resolution (as well as any other channels that are in a user's .condarc file).  Currently it is not possible to specify a specific channel/package combination, rather the dependency resolver chooses the source for each package from the specified channels.

jsignell, 2016-09-09:
Is this still in the works? I would be happy to take a look at it if other people aren't. @scollis in terms of "will this make it harder to install" maybe it could be that Basemap and Cartopy are just two different options for mapping. So if Cartopy isn't installed the mapping could just fall back to Basemap.

jjhelmus, 2016-09-12:
@jsignell This feature been proposed but as far as I know no-one has stepped up to champion the implementation.  A few weeks back I wrote a quick proof of concept Cartopy based **RadarMapDisplayCartopy** class with the idea that it could be a replacement for the basemap based **RadarMapDisplay**.  This lives in the [cartopy_plot branch](https://github.com/jjhelmus/pyart/commits/cartopy_plot) in my fork of Py-ART.  If anyone wanted to pick this work up I would welcome it.

jsignell, 2016-09-26:
@jjhelmus I am taking a look at it this week. I'll let you know how it goes.

jsignell, 2016-09-26:
@jjhelmus I made some progress with the cartopy implementation. I was trying to replicate the test figure from RadarMapDisplay.plot_ppi which looks like this:

[image]

This is the Cartopy version:
[image]

I implemented the other tests that RadarMapDisplay has, but I would be interested in knowing what pitfalls you foresee. I don't know where to submit a pull request, but I guess you can check out my work on [my fork]( https://github.com/jsignell/pyart/tree/cartopy-plot).

jjhelmus, 2016-10-12:
@jsignell Very nice work on the additional implementation.  

It looks like a Lambert Conformal projection is always set as the `proj` attribute.  This really should be projection specified in the `plot_ppi_map` method call.  This might add some difficulty as this attribute is used in a number of other methods and I'm not sure if other projections have similar behavior.  

There are a few other changes I would recommend but those can address when you submit this as a PR.

scollis, 2016-10-12:
We may also be able to task @Rumpkie to assist @jsignell in crafting a Pull Request. Since this is an identified need in the draft roadmap and @Rumpkie could use some practice I think it is a good use of time.. He could also task @zssherman as well

jsignell, 2016-10-17:
@jjhelmus yeah I was struggling with how to deal with the projections. I don't see where the `proj` in `RadarMapDisplay` is ever used. I was using the projection in `plot_ppi_map` as a kwarg that could be used to define the desired output projection. And I was using proj to keep track of the original projection of the data.

swnesbitt, 2016-10-18:
This is great, thanks Julia!

jsignell, 2016-10-18:
I opened a PR for this #612

nguy, 2017-04-11:
Can this issue be closed? Have any problems been reported with using the cartopy mapping?

I need to try this.

scollis, 2017-04-11:
I don't want to close it yet.. we have cartopy for ppis.. need it for grid objects 

And yes. I am getting happier with it and Metpy compatibility is awesome 

-sent from a mobile device-

zssherman, 2020-03-26:
Was added to grid objects, closing.

closed 2020-03-26 by zssherman