ARM-DOE/pyart #484: ODIMH5 PPI orientiation bug

joshuass, 2016-02-11:
Hi,

Just an easy fix bug. The new version of py-ART (1.6) shows PPIs rotated 90 degrees anticlockwise. I had no issues in v1.5 with the same datasets.

[Link to ODIMH5 file](https://www.dropbox.com/s/s7dygcipt5ti15o/201512152330Kurnell.h5?dl=0)

[Link to ipynb file](https://github.com/uq-crg/rapic_to_pyart/blob/master/Kurnell_tor_20151215.ipynb)

Cheers,
J

jjhelmus, 2016-02-11:
@joshuass can you elaborate a bit on the bug you are seeing?  When I plot the seventh tilt from the linked ODIM file I get the same results with Py-ART 1.5.0, 1.6.0 and the currently master:

## Py-ART 1.5.0:
[image]

## Py-ART 1.6.0:
[image]

## Py-ART 1.7.0.dev

[image]

joshuass, 2016-02-12:
Hi @jjhelmus 

I've done some more investigation and found the source of this issue

@scollis  ipynb of the event used odim_h5 data generated by an older converter where the dataset#/data#/where/a1gate variable was always set to 0. The data is oriented correctly in this case.
notebook: https://github.com/scollis/notebooks/blob/master/Kurnell.ipynb
old odim file: https://www.dropbox.com/s/i5afnrlbdoskzmk/old_201512152330Kurnell.h5?dl=0

I copied his notebook and used the new converter, where the a1gate is now correctly set to the starting azimuth in the scan (248 in this case). Data is rotated CC ~90deg
notebook: https://github.com/uq-crg/rapic_to_pyart/blob/master/Kurnell_tor_20151215.ipynb
data: https://www.dropbox.com/s/s7dygcipt5ti15o/201512152330Kurnell.h5?dl=0

The datasets within these two files have exactly the same specification according to the ODIM standard 2.2, e.g., the northern most ray is first regardless of the a1gate. I suspect py-ART is reading the a1gate and offsetting the data azimuth?

Let me know what you think!

joshuass, 2016-02-12:
Found this issue in the py-ART code.
https://github.com/ARM-DOE/pyart/blob/master/pyart/aux_io/odim_h5.py#L218

the a1gate isn't used to orientate the data in odimh5 files according to the specifications, the northward most ray is always the first row.

See section 5.1 of...
http://www.eumetnet.eu/sites/default/files/OPERA2014_O4_ODIM_H5-v2.2.pdf

jjhelmus, 2016-02-12:
Agreed.  There are a number of improvements and bugs that need to be fixed in Py-ART's odim h5 reader, #445.  If you want to take a shot at fixing the code up a bit @joshuass feel free. I have it on my to-do list but I can't give a timeline as to when I will get to it.

joshuass, 2016-02-18:
I'll have a go at this small bug @jjhelmus , but I'm afraid my python skills are a little lacking to review the entire ODIM reader. Once I finish my PhD in a few months I'll have plenty of time to perfect these skills and contribute more.

Also, you'll be interested to hear we've started building a complete volumetric archive of Australia radar data in odim for open access to researchers. I've introduced py-art to the Australian community with a number of notebooks Scott wrote and the response has been very promising. Hopefully py-art will be the tool of choice in Australia soon enough :)

jjhelmus, 2016-02-22:
@joshuass I'm excited to head about the creation of an Australian radar archive, would be a great resource for the community.  Let me or Scott know if you need any materials to introduce or show off Py-ART.  

I'm traveling this week and may have a some time on the plane or in the evening to improve Py-ART ODIM support.  I'll keep this issue updated with any progress.

joshuass, 2016-02-23:
Cheers! It would be great to host a few notebooks of select cases (Scott already has enough written) with some specific documentation for odim files. Not much really. I've given it a start here [link](https://github.com/uq-crg/rapic_to_pyart/wiki/Using-rapic_to_odim) , but happy to move everything onto a DOE page. I'll raise it as a request once the archive is finished.

closed 2016-06-13 by jjhelmus