ARM-DOE/pyart #487: writing cfradial changes attributes to strings, matlab cannot read this

nphoban, 2016-02-17:
I am attempting to read in a cfradial file, dealias the velocities, and output a new cfradial file with the added dealiased velocity field. Along the way, the attributes of the existing data become changed to a "string" format. This string format is not compatible with matlab and I need it to be. The added dealiased velocity field is outputted correctly, but all the other fields are changed. I have attached a picture of my cfradial file before and after the processing. 

I believe this issue might be from line 700 in cfradial.py. Here data that is type 'U' is assigned data type 'S'. 
[image]
[image]

jjhelmus, 2016-02-17:
@nphoban Can you provide a bit more details on this issue?  I can't see from the screenshots posted what variables are changing.  

One though is that by default the `pyart.io.write_cfradial` function creates netCDF4 version 4 files which may not be able to be read in Matlab.  If you set the **format** parameter to 'NETCDF3_CLASSIC' then the function will create a netCDF version 3 files.

scollis, 2016-02-17:
> Jonathan J. Helmus <mailto:notifications@github.com>
> February 17, 2016 at 5:20 PM
> nphoban <mailto:notifications@github.com>
> February 17, 2016 at 2:59 PM
> beforepythonprocessing
> afterpythonprocessing

Is it that it is now writing strings rather than an array of Chars?

nphoban, 2016-02-18:
@jjhelmus Matlab can read netCDF4 and I have also tried writing netCDF3 using 'NETCDF3_CLASSIC' and that did not fix the problem. 

In the images, if you look below the float variables you will see the word string is now added in front of most of the variables attributes. It appears the actual data in the variables is fine, but the attributes have changed to a format that matlab cannot read. Except for the added variable dealiased_velocity where the attributes and data are written correctly.

jjhelmus, 2016-02-18:
I understand the issue now, thanks for clarifying.  I was looking at the wrong information in the images, sorry about that.  I'm not sure what exactly is causing this but I will look into it and get back when I have an answer.

jjhelmus, 2016-02-18:
From the testing I've done, this conversion of text/byte attribute to string attributes seems to be caused by a recent update to the netcdf4-python library.  You may be able to work around this issue by installing an older version of netcdf4 which does not write out string attributes.  Version 1.1.1 should work and can be installed using `conda install netcdf4==1.1.1` if you are using Python.

I've filed an issue with the netcdf4-python library (Unidata/netcdf4-python#529) to see if they have any recommendation on how to deal with this issue.

jjhelmus, 2016-02-18:
xarray seems to be discussing how to deal with the same issue, pydata/xarray#681

jjhelmus, 2016-02-22:
I'm still working with the netcdf4-python developers on a good solution to this issue.  For the time being my suggestion would be to use an older version of the netcdf4-python library if you need to create NetCDF file with Py-ART that can be read in Matlab.

nphoban, 2016-02-22:
Thank you for your help.

jjhelmus, 2016-08-17:
I believe this issue has been address in changes to netcdf4-python in the linked issue, closing.

If you are still running into this issue @nphoban or anyone else feel re-open this issue.

closed 2016-08-17 by jjhelmus