ARM-DOE/pyart #492: 3d visualization of radar data using a GIS viewer.

Rumpkie, 2016-03-02:
A useful new feature for rendering radar data onto a map would be to create pyart io output that would be compatible with GIS toolkits like the webgl virtual planet Cesiumjs from http://cesiumjs.org/
Here is a netCDF grid http://cesiumjs.org/demos/GridViz.html example:
[image]

another example… something like an RHI plot. 
http://abhishekvpotnis.me/data-curtains/
[image]

Could be useful, would definitely be cool.

How it would work:

Pyart io would write the selected and processed radar object into a GEOJSON file similar to the process that radar display uses to get the data ready for matplotlib.

Cesium would be used to display interact with the data, and jquery would be used to add view controls.

jjhelmus, 2016-03-02:
This would be interesting, although I think would require a decent amount of work.  I know there has been some work to output geo-referenced raster images of radar fields using Py-ART for GIS but I don't think GeoJSON has been suggested before.

scollis, 2017-07-13:
I see no obvious path to a solution in the near future. I am closing. If there is interest open a new issue or re-open this.

closed 2017-07-13 by scollis

AdamTheisen, 2021-09-22:
@scollis @zssherman was the ability to export in GeoJSON ever implemented?  Was thinking it would make for an interesting and interactive way to plot and monitor radar data using plotly chloropleth  plots.

zssherman, 2021-09-22:
@AdamTheisen sadly it has not, the closest thing to Pyart and 3d is yt project PR, but they haven't made progress in a bit.