ARM-DOE/pyart #496: CyLP and CVXOPT unit tests skipped on Travis CI

jjhelmus, 2016-03-07:
Currently Travis CI is skipping unit tests which depend on cylp or cvxopt, this should be fixed.  For example [this build](https://travis-ci.org/ARM-DOE/pyart/jobs/114306856).

jjhelmus, 2016-03-07:
The issue with CyLP and CVXOPT is that both packages have Python extensions that link to liblapack and libblas which depend on libgfortran.  The symbols provided by the conda libgfortran libraries are not sufficient for the lapack and blas libraries installed on the system.  Force removing this package results in the system libgfortran library being used.  

A more preferable solution to this would be to not use the system provided liblapack or libblas at all and instead meet these dependencies with conda packages which are linked to the conda provided libgfortran.  This will take time, I will look into it when I move the cylp and cxvopt_glpk recipes over to conda-forge.

scollis, 2016-03-07:
Sounds reasonable

PR #497 (2016-03-07): TST: Use system libgfortran on Travis CI
closes #496
commit 9babe8ee (2016-03-07): TST: Use system libgfortran on Travis CI

closes #496
closed 2016-03-07 by jjhelmus