ARM-DOE/pyart #499: Can't read old (2012) NEXRAD files

jacaicedob, 2016-03-10:
I get the following error when I try to read the .gz file:
"ValueError: Gzip file cannot be read compressed, uncompress and try again"

I unzipped the file and got the following error:
"TypeError: Unknown or unsupported file format: UNKNOWN"

The files I am trying to open can be found here:
http://www.ncdc.noaa.gov/nexradinv/bdp-download.jsp?id=KJAX&yyyy=2012&mm=07&dd=17&product=AAL2

Specifically, I was looking for times between 1800 and 2200 UTC.

Thanks

scollis, 2016-03-10:
I thought @jjhelmus fixed this.. You can use http://www.eol.ucar.edu/software/radx to do a convert to CF-Radial for now.. I had to do this for a project based out of Maine as well. 
here is my param file 
[scollis@blogin1 ~]$ more /homes/scollis/radxconvert_oldnexrad.params 
set_output_fields = TRUE;

///////////// output_fields ///////////////////////////
//
// Output field details.
// Set the details for the output fields. The output_field_name is the 
//   ndtCDF variable name. Set the long name to a more descriptive name. 
//   Set the standard name to the CF standard name for this field.
//
// Type: struct
//   typedef struct {
//      string input_field_name;
//      string output_field_name;
//      string long_name;
//      string standard_name;
//      string output_units;
//      output_encoding_t encoding;
//        Options:
//          OUTPUT_ENCODING_ASIS
//          OUTPUT_ENCODING_FLOAT32
//          OUTPUT_ENCODING_INT32
//          OUTPUT_ENCODING_INT16
//          OUTPUT_ENCODING_INT08
//   }
//
// 1D array - variable length.
//

output_fields = {
  { "REF", "reflectivity", "reflectivity", "equivalent_reflectivity_factor", "dBZ", OUTPUT_ENCODING_ASIS},
  { "VEL", "velocity", "radial_velocity", "radial_velocity_of_scatterers_away_from_instrument", "m/s", OUTPUT_ENCODING_ASIS},
  { "SW", "spectrum_width", "spectrum_width", "spectrum_width", "m/s", OUTPUT_ENCODING_ASIS},
  { "RHO", "copol_coeff", "cross_correlation_ratio_hv", "cross_correlation_ratio_hv", "ratio", OUTPUT_ENCODING_ASIS},
  { "PHI", "dp_phase_shift", "differential_phase_hv", "differential_phase_hv", "degrees", OUTPUT_ENCODING_ASIS},
  { "ZDR", "diff_reflectivity", "log_differential_reflectivity_hv", "log_differential_reflectivity_hv", "dB", OUTPUT_ENCODING_ASIS}
};

jacaicedob, 2016-03-10:
@scollis I have never used Radx before. What parameters do you pass to RadxConvert?

I tried running it as follows:
./RadxConvert -params radxconvert_oldnexrad.params file_to_convert

and it gives me a lot of ">>> TDRP_WARNING <<< - struct parameter 'output_fields'
" warnings and halts.

scollis, 2016-03-10:
Warnings are ok.
here is how I run it:
~/radxbin/bin/RadxConvert -params 
/homes/scollis/radxconvert_oldnexrad.params -f 
/lcrc/group/earthscience/radar/stage/www1.ncdc.noaa.gov/pub/has/HAS010695119/KGYX200612*

jacaicedob, 2016-03-10:
@jjhelmus you may be right. I am getting a lot of "File format not recognized:" errors with RadxConvert even on the 40 MB ones.

swnesbitt, 2016-03-10:
They are also reported as corrupt on the UNIDATA THREDDS server:

```
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-2-2dcc4621954d> in <module>()
     25 #http://thredds-aws.unidata.ucar.edu/thredds/fileServer/nexrad/level2/S3/2016/02/05/KOKX/KOKX20160205_120102_V06.gz
     26 #http://thredds-aws.unidata.ucar.edu/thredds/dodsC/nexrad/level2/S3/2016/02/05/KOKX/KOKX20160205_120102_V06.gz
---> 27 radar = pyart.io.read_nexrad_archive(filename,linear_interp=True)
     28 #import os
     29 import shutil

/data/keeling/a/snesbitt/anaconda2/envs/py3k/lib/python3.5/site-packages/pyart/io/nexrad_archive.py in read_nexrad_archive(filename, field_names, additional_metadata, file_field_names, exclude_fields, delay_field_loading, station, scans, linear_interp, **kwargs)
    109 
    110     # open the file and retrieve scan information
--> 111     nfile = NEXRADLevel2File(prepare_for_read(filename))
    112     scan_info = nfile.scan_info(scans)
    113 

/data/keeling/a/snesbitt/anaconda2/envs/py3k/lib/python3.5/site-packages/pyart/io/nexrad_level2.py in __init__(self, filename)
    142             buf = fh.read()
    143         else:
--> 144             raise IOError('unknown compression record')
    145         self._fh = fh
    146 

OSError: unknown compression record
```

jacaicedob, 2016-03-10:
Thanks to all of you for your input.

closed 2016-03-10 by jacaicedob