ARM-DOE/pyart #500: Unknown or unsupported file format

candidobugarin, 2016-03-12:
Hello.

I'm trying to open a file from a band C radar But when I try to open the file, python return a message " TypeError : unknown file format or unsupported : UNKNOWN ." you can open the file? the file extension is ".volume".
Thanks

swnesbitt, 2016-03-13:
Can you post the file for download?  Which radar is this from?

candidobugarin, 2016-03-13:
This is the link to the file.
https://drive.google.com/file/d/0B17oNKCpatJSQmkxOU9JMjRxT00/view
It is a radar Band-C located in Rio de Janeiro

swnesbitt, 2016-03-13:
I tried reading it with Radx, RSL, and Py-ART with no luck.  I also tried to read it as a GIS shapefile, but there needs to be other files with it to (i.e., shx).  I would contact the data provider to get some more information. There is some radar information in the binary file (`SWEEP`).

```
(py3k)granizo:Downloads snesbitt$ xxd shpp.volume.shp | head
0000000: 6473 6870 0000 3c00 6c2b 0800 31bd feff  dshp..<.l+..1...
0000010: 2e60 0200 4c51 fe54 205f fe54 0001 0004  .`..LQ.T _.T....
0000020: 4331 0700 0300 0000 5357 4545 5000 0000  C1......SWEEP...
0000030: 4c51 fe54 dc17 0000 3822 0000 ee02 1500  LQ.T....8"......
0000040: 0100 0000 4385 6001 5f82 4881 6582 6261  ....C.`._.H.e.ba
0000050: 4363 6001 4601 6002 4581 6348 6242 6262  Cc`.F.`.E.cHbBbb
0000060: 8262 4280 0169 0160 0245 8242 6143 8441  .bB..i.`.E.BaC.A
0000070: 8241 6145 8001 6d01 6002 4669 4162 8480  .AaE..m.`.FiAb..
0000080: 0172 0160 0247 6146 8242 8662 8241 8260  .r.`.GaF.B.b.A.`
0000090: 0149 0160 0248 6145 6261 4182 4162 4281  .I.`.HaEbaA.AbB.
```

scollis, 2016-03-13:
So is the file off the radar? Where did you get the file from? Who provided it? Without this info we are shooting in the dark 

-sent from a mobile device-

jjhelmus, 2016-03-13:
@candidobugarin If you can provide additional details on where this file came from we may be able to figure out a method of reading it but it does not look as if Py-ART supports the file format directly..

candidobugarin, 2016-03-13:
I also have little information about the file. Who gave me the file was one of the people working with radar. From what I understand , is a radar band C used for televisions in the US. Your software can not convert the volume netcdf or hdf5 . The only way to get the data is through this file or a png image. Tomorrow I can tell the person who gave me the file and get more information for you.

swnesbitt, 2016-03-13:
@candidobugarin OK, it would be good to know which radar you are looking at.  The Sumaré radar page on the [WMO radar database](http://wrd.mgm.gov.tr/db/radar-details.aspx?l=en&r=2975) says it was manufactured by ADC.  I tried to google their radar format without any luck, perhaps the radar itself has software to export the data to netCDF?  We would need to know how this binary data is structured to create a reader.

jjhelmus, 2016-03-15:
@candidobugarin Any additional information about this file?  Without something to work from I do not think we can guess the format of this file.

jjhelmus, 2016-03-23:
Since the original poster has not gotten back to us I'm going to close this for the time being.  

@candidobugarin If you are still having issue and can provide additional details feel free to re-open this issue.

closed 2016-03-23 by jjhelmus

candidobugarin, 2016-03-25:
Hello.

Sorry I am late. They sent me today the radar software manual. The radar in question is the Sumaré Radar, operated by the City of Rio de Janeiro . The link to manual is: https://drive.google.com/file/d/0B17oNKCpatJSTDVIenZMX2ZoZW8/view

Thanks

jjhelmus, 2016-03-28:
Unfortunately, Py-ART does not support reading files from ADC DOPRAD radars and I am not familiar with any software which is able to covert these files into a format that Py-ART does support.